android/camera-samples #1: How can we take the picture and convert it to base64?

jaypratapsingh, 2019-05-08:
I want to click the picture using camera and convert into base64.

Pinkal7600, 2019-05-09:
@jaypratapsingh To convert image into base64 you can check solution from StackOverflow.

jaypratapsingh, 2019-05-09:
No, I mean I want to get the base64 when we take the picture from the camera.

pwittchen, 2019-05-09:
I think, it's not in the scope of this project. Once, you capture the picture as a `File` type, then you can do it whatever you want (e.g. convert it into base64 format).

Here you have simple code snippet in Java with using Apache Commons IO:

```
    File file = new File(...);

    imageCapture.takePicture(file,
        new ImageCapture.OnImageSavedListener() {
            @Override
            public void onImageSaved(File file) {
                byte[] fileContent = FileUtils.readFileToByteArray(file);
                String encodedString = Base64.getEncoder().encodeToString(fileContent);
            }
            @Override
            public void onError(
                    ImageCapture.UseCaseError useCaseError,
                    String message,
                    Throwable cause) {
                // handle error here
            }
    }

```

It's just one example. I think, in case of Android you can try to search for a more lightweight solution for conversion without additional library to the reduce the final app size. Alternatively, you can try to copy code of this method into your project if license allows to do that, because AFAIK, Apache Commons IO is open-source project.

Here you have reference articles:
- https://developer.android.com/training/camerax/take-photo#java
- https://www.baeldung.com/java-base64-image-string

jaypratapsingh, 2019-05-10:
Thanks, @pwittchen, yup I can go with it meanwhile if there is any other way by which I could get directly base64 it would be great.

owahltinez, 2019-05-14:
@jaypratapsingh unfortunately capturing images directly  in base64 is out of scope for this project as well as the library itself.

closed 2019-05-14 by owahltinez