android/camera-samples #2: Record video

Taharo, 2019-05-08:
Does CameraX support video record ?

Pinkal7600, 2019-05-09:
Yes, we can record video using CameraX. I have done but I don't know its correct or not. Please refer below code may it helps you.

**Config for Video in CameraX:**

        val videoCaptureConfig = VideoCaptureConfig.Builder().apply {
            setLensFacing(lensFacing)
            setTargetAspectRatio(screenAspectRatio)
            setTargetRotation(viewFinder.display.rotation)

        }.build()

        videoCapture = VideoCapture(videoCaptureConfig)

        CameraX.bindToLifecycle(this, preview, imageCapture, videoCapture)

**To Start video recording:**

        videoCapture?.startRecording(videoFile, object : VideoCapture.OnVideoSavedListener {
                override fun onVideoSaved(file: File?) {
                    Log.i(javaClass.simpleName, "Video File : $file")
                }

                override fun onError(useCaseError: VideoCapture.UseCaseError?, message: String?, cause: Throwable?) {
                    Log.i(javaClass.simpleName, "Video Error: $message")
                }

            })

**To Stop video recording:**

        videoCapture?.stopRecording()

Taharo, 2019-05-09:
Thanks for your response. But not all works fine.

`
CameraX.bindToLifecycle(this, preview, videoCapture, imageAnalyzer)
`
lead to 
`java.lang.ArithmeticException: divide by zero`

owahltinez, 2019-05-14:
@Taharo video recording is currently in preliminary stages and not officially supported -- but we are working on it! Please see this related [StackOverflow question](https://stackoverflow.com/a/56055930/440780).

closed 2019-05-14 by owahltinez

sandeep-sporttotal, 2019-10-28:
I have built small app which uses CameraX videocapture App crashes when pressed back while recording is on progress. Need some help to fix this.
Dependencies : camerax:camera-core 1.0.0-alpha06 & camerax:camera-camera2 : 1.0.0-alpha06

```
FATAL EXCEPTION: CameraX-video encoding thread
    Process: <packagename>, PID: 10794
    java.lang.NullPointerException: Attempt to invoke virtual method 'int android.media.MediaCodec.dequeueOutputBuffer(android.media.MediaCodec$BufferInfo, long)' on a null object reference
        at androidx.camera.core.VideoCapture.videoEncode(VideoCapture.java:604)
        at androidx.camera.core.VideoCapture$2.run(VideoCapture.java:348)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)

```

sandeep-borndigital, 2019-10-28:
I have built small app which uses CameraX videocapture App crashes when pressed back while recording is on progress. Need some help to fix this.
Dependencies : camerax:camera-core 1.0.0-alpha06 & camerax:camera-camera2 : 1.0.0-alpha06

FATAL EXCEPTION: CameraX-video encoding thread
    Process: <packagename>, PID: 10794
    java.lang.NullPointerException: Attempt to invoke virtual method 'int android.media.MediaCodec.dequeueOutputBuffer(android.media.MediaCodec$BufferInfo, long)' on a null object reference
        at androidx.camera.core.VideoCapture.videoEncode(VideoCapture.java:604)
        at androidx.camera.core.VideoCapture$2.run(VideoCapture.java:348)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)

RamapatiMaurya, 2019-12-07:
We have found a CameraX sample on Github https://github.com/robertlevonyan/CameraXDemo. By using this library we can capture photos and videos. While customizing this sample code i am not able to find any way to pause the video recording even start and stop video recording feature is there.

Any help will be appreciated!!!

williamycyh, 2019-12-19:
> I have built small app which uses CameraX videocapture.
> 
> Issue : App crashes when pressed back while recording is on progress. Need some help to fix this.
> Dependencies : camerax:camera-core 1.0.0-alpha06 & camerax:camera-camera2 : 1.0.0-alpha06
> 
> ```
> FATAL EXCEPTION: CameraX-video encoding thread
>     Process: <packagename>, PID: 10794
>     java.lang.NullPointerException: Attempt to invoke virtual method 'int android.media.MediaCodec.dequeueOutputBuffer(android.media.MediaCodec$BufferInfo, long)' on a null object reference
>         at androidx.camera.core.VideoCapture.videoEncode(VideoCapture.java:604)
>         at androidx.camera.core.VideoCapture$2.run(VideoCapture.java:348)
>         at android.os.Handler.handleCallback(Handler.java:873)
>         at android.os.Handler.dispatchMessage(Handler.java:99)
>         at android.os.Looper.loop(Looper.java:193)
>         at android.os.HandlerThread.run(HandlerThread.java:65)
> ```

I also got this error, how to resolve

vick16ach, 2020-02-07:
I wonder what kind of people in google, develop such a feature without proper documentation.
I'm getting this error and no way i can find the solution.

```
    java.lang.NullPointerException: Attempt to invoke virtual method 'void android.media.AudioRecord.startRecording()' on a null object reference
```

Sowmya-Flickr, 2020-03-19:
When is Video record api expected to be released? It not part of the beta release, correct?

saurabhthorat, 2020-05-14:
@Sowmya-Flickr According to their [IssueTracker](https://issuetracker.google.com/issues?q=componentid:618491), they would start looking into video capture after they finish up their beta release.

awscherb, 2020-07-14:
@vick16ach make sure you request `RECORD_AUDIO` permission

ViduraPrasangana, 2020-08-02:
> I wonder why people in google, develop such a feature without proper documentation. I thought google has a well development strategies in the world.
> I'm getting this error and no way i can find the solution.
> 
> ```
>     java.lang.NullPointerException: Attempt to invoke virtual method 'void android.media.AudioRecord.startRecording()' on a null object reference
> ```

Adding `android.permission.RECORD_AUDIO` to the permissions will fix this

NaveenKadiyala, 2020-08-08:
> I have built small app which uses CameraX videocapture.
>
> Issue : App crashes when pressed back while recording is on progress. Need some help to fix this.
> Dependencies : camerax:camera-core [...]

I faced the same issue, i applied some hack to reslove this, it worked but not sure whether to follow this.
Just  overrided the onBackPressed of activity and stopped recording `videoCapture.stopRecording();`.
Stopping recording will call the onVideoSaved() and i was quitting the activity from there.
By using some boolean values you can acheive this.. hope this helps...

kenmasumitsu, 2020-08-13:
Are there any ways to record video without audio ?
I tried `VideoCapture.Builder.setAudioBitRate(0)`, but not audio is included in the video.

idish, 2020-10-08:
> Adding `android.permission.RECORD_AUDIO` to the permissions will fix this

It doesn't fix it.

Dacen12, 2021-01-29:
> 
> 
> @Taharo video recording is currently in preliminary stages and not officially supported -- but we are working on it! Please see this related [StackOverflow question](https://stackoverflow.com/a/56055930/440780).

how is it then possible that snapchat and whatsapp and instagram use video recording services without problems?

darrinps, 2021-03-03:
> how is it then possible that snapchat and whatsapp and instagram use video recording services without problems?

I think he was referring to CameraX's support for it. It can be done in CameraX now though and from what I've seen, works better than Camera2 in many ways. Not everything is there though. No way to add a watermark/overlay like in iOS. No way (that I can find) to shut off audio recording. Etc.

DeluxeViper, 2021-09-10:
For anyone looking to solve the null pointer exception, you must add the RECORD_AUDO permission to Android Manifest:

`<uses-permission android:name="android.permission.RECORD_AUDIO"/>`