android/camera-samples #3: How to zoom Camera?

Pinkal7600, 2019-05-09:
I have tried to zoom camera with below code. Please check and correct me where I am wrong.

First I get a **Preview** from **CameraX** and tried to perform zoom using **Preview** like below.

       var config = CameraX.getDefaultUseCaseConfig(PreviewConfig::class.java, lensFacing)
       var preview = Preview(config)
       preview.zoom(zoom)

After preview.zoom() I just bind again with CameraX and get some error and it's not working. 

       CameraX.bindToLifecycle(this, preview, imageCapture, videoCapture)

When the above code was not working, I also tried with **CameraX.unbindAll()** first and then I call **CameraX.bindToLifecycle()**. But face some error again and not get succeed.

Please provide an answer on below question or add some description in Wiki or Readme file related to the below question. 

1. How we can zoom camera? 
2. How to get max zoom from **CameraX**?

And also if possible please provide how to manage flash with **CameraX**?

Thank you!

owahltinez, 2019-05-14:
This issue is unrelated to the sample, for issues related to the library please use StackOverflow using the `android-camerax` tag or the relevant [component in Android's issue tracker](https://issuetracker.google.com/issues?q=componentid:618491).

To answer your question, take a look at https://developer.android.com/reference/androidx/camera/core/Preview.html#zoom(android.graphics.Rect). The Rect passed as a parameter should correspond to what you would set for [Camera2's crop region parameter](https://developer.android.com/reference/android/hardware/camera2/CaptureRequest#SCALER_CROP_REGION).

owahltinez, 2019-09-04:
Please see the [CameraControl](https://developer.android.com/reference/androidx/camera/core/CameraControl) class which will implement methods for zoom

closed 2019-09-04 by owahltinez

ktchernov, 2019-10-24:
@owahltinez the Preview#zoom documentation does not answer either of the two questions, which would be quite common.

The Android issue tracker is for logging bugs, not for asking questions.

It would be good for the official sample project to demonstrate what is quite common functionality, which is not otherwise sufficiently covered in the documentation.

Pinkal7600, 2020-01-28:
Zoom API is now available from CameraX Team.
Please check my answer for more details. 
StackOverflow: https://stackoverflow.com/a/59855737/7235539