android/camera-samples #4: How to specify video resolution for VideoCapture ?

Taharo, 2019-05-09:
I want to set video resolution width=540 and height=960, but resulted video has width=360 and height=640.

`
val videoCaptureConfig = VideoCaptureConfig.Builder().apply {
            setLensFacing(lensFacing)
            setTargetRotation(viewFinder.display.rotation)
            setTargetResolution(Size(540, 960))
            setMaxResolution(Size(540, 960))
        }.build()

videoCapture = VideoCapture(videoCaptureConfig)

CameraX.bindToLifecycle(this, preview, videoCapture)
`

josephteoh, 2019-05-10:
@Taharo, may I know how did you able to use VideoCapture to capture video? I tried but got an error "VideoCapture class have annotation RestrictTo(Scope.LIBRARY_GROUP)"...

Taharo, 2019-05-10:
@josephteoh just add @SuppressLint("RestrictedApi") on your method.
`
@SuppressLint("RestrictedApi")
    override fun onViewCreated(view: View, savedInstanceState: Bundle?) {
        super.onViewCreated(view, savedInstanceState)
}
`

josephteoh, 2019-05-11:
@Taharo thx, added it, but encountered other error with regards to AudioRecord....guess it's still buggy(in Java)...probably I should try Kotlin then. Thx again, sorry for the sidetrack and unable to test out your question.

owahltinez, 2019-05-14:
@Taharo video recording is currently in preliminary stages and not officially supported, buggy and inconsistent behavior like you described here is expected at this stage.

closed 2019-05-14 by owahltinez