android/camera-samples #6: Should TextureView be released when preview use case is no more used?

nominalista, 2019-05-09:
I found in the `Preview` class documentation:

> It is the application's responsibility to release the last SurfaceTexture returned by Preview.PreviewOutput.getSurfaceTexture() when a new SurfaceTexture is provided via an update or when the user is finished with the use case.

Does it mean that in `onDestroyView` along with `unbindAll`, the `SurfaceTexture` of `viewFinder` shouldn't be released?

owahltinez, 2019-05-14:
@nominalista releasing the SurfaceTexture is unnecessary in `onDestroyView` because all children views should get destroyed with it.

nominalista, 2019-05-15:
@owahltinez clear. Thanks for explanation! This issue can be closed I think.

closed 2019-05-15 by nominalista