android/camera-samples #8: App crashes after switching cameras multiple times

nominalista, 2019-05-10:
When you try to switch cameras by clicking really fast in a switch button, the app crashes due to this exception:

```
E/Legacy-CameraDevice-JNI: getNativeWindow: Surface had no valid native window.
E/Legacy-CameraDevice-JNI: LegacyCameraDevice_nativeDetectSurfaceDimens: Could not retrieve native window from surface.
E/AndroidRuntime: FATAL EXCEPTION: CameraX-
    Process: com.android.example.cameraxbasic, PID: 11109
    java.lang.IllegalArgumentException: Surface was abandoned
        at android.hardware.camera2.utils.SurfaceUtils.getSurfaceSize(SurfaceUtils.java:84)
        at android.hardware.camera2.params.OutputConfiguration.<init>(OutputConfiguration.java:260)
        at android.hardware.camera2.params.OutputConfiguration.<init>(OutputConfiguration.java:145)
        at android.hardware.camera2.impl.CameraDeviceImpl.createCaptureSession(CameraDeviceImpl.java:508)
        at androidx.camera.camera2.impl.CaptureSession.open(CaptureSession.java:196)
        at androidx.camera.camera2.impl.Camera.openCaptureSession(Camera.java:535)
        at androidx.camera.camera2.impl.Camera$StateCallback.onOpened(Camera.java:743)
        at androidx.camera.core.CameraDeviceStateCallbacks$ComboDeviceStateCallback.onOpened(CameraDeviceStateCallbacks.java:99)
        at android.hardware.camera2.impl.CameraDeviceImpl$1.run(CameraDeviceImpl.java:143)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
     Caused by: android.hardware.camera2.legacy.LegacyExceptionUtils$BufferQueueAbandonedException
        at android.hardware.camera2.legacy.LegacyExceptionUtils.throwOnError(LegacyExceptionUtils.java:73)
        at android.hardware.camera2.legacy.LegacyCameraDevice.getSurfaceSize(LegacyCameraDevice.java:606)
        at android.hardware.camera2.utils.SurfaceUtils.getSurfaceSize(SurfaceUtils.java:82)
        at android.hardware.camera2.params.OutputConfiguration.<init>(OutputConfiguration.java:260) 
        at android.hardware.camera2.params.OutputConfiguration.<init>(OutputConfiguration.java:145) 
        at android.hardware.camera2.impl.CameraDeviceImpl.createCaptureSession(CameraDeviceImpl.java:508) 
        at androidx.camera.camera2.impl.CaptureSession.open(CaptureSession.java:196) 
        at androidx.camera.camera2.impl.Camera.openCaptureSession(Camera.java:535) 
        at androidx.camera.camera2.impl.Camera$StateCallback.onOpened(Camera.java:743) 
        at androidx.camera.core.CameraDeviceStateCallbacks$ComboDeviceStateCallback.onOpened(CameraDeviceStateCallbacks.java:99) 
        at android.hardware.camera2.impl.CameraDeviceImpl$1.run(CameraDeviceImpl.java:143) 
        at android.os.Handler.handleCallback(Handler.java:873) 
        at android.os.Handler.dispatchMessage(Handler.java:99) 
        at android.os.Looper.loop(Looper.java:193) 
        at android.os.HandlerThread.run(HandlerThread.java:65) 
```

To reproduce this issue just run the app and tap switch button multiple times (like 15-20). I know this is not production ready app and tapping so many times fakes user behavior, but I have encountered the same issue in my own implementation just because of fragment replacing. It would be much appreciated if someone give a clue or some explanation of this issue.

owahltinez, 2019-05-14:
@nominalista what device are you experiencing this in?

nominalista, 2019-05-15:
@owahltinez Pixel 3.

owahltinez, 2019-05-25:
Unfortunately this is an error happening inside of CameraX which cannot be (easily) caught by the app. If I'm reading the logs correctly, it could be caused by the camera session sending frames to a surface that has been recycled. 

Based on the behavior that you described, a workaround could be to prevent users from switching fragments until the session has been properly started. For example, keeping the button disabled until preview starts.

I'll work on getting a reliable way to reproduce this on our side, and see if we can add the workaround to the sample as well.

owahltinez, 2019-05-29:
@nominalista I am not able to reproduce this issue on a Pixel 3 XL. I tried pressing the switch button 15-20 times very fast, not giving the preview enough time to load; then a little slower, immediately after preview is shown; and then even slower, every 2-3 seconds.

Unfortunately I have been unable to test on the emulator due to #9

owahltinez, 2019-06-07:
Now that #9 is fixed I tested on the emulator but I'm also unable to reproduce. Are you still able to reproduce this issue? Can you please try again with the updated sample and, if it still crashes, share what OS build your Pixel 3 is running?

nominalista, 2019-06-07:
Sorry for late response, [this video shows the issue](https://imgur.com/9QbPXBa). 
After update everything works fine. Thanks and I think this issue can be closed now.

closed 2019-06-07 by nominalista

franciscerio, 2019-07-03:
@nominalista I think the camera is working fine. You can add some blocker/stopper when he pressed 300-400 milliseconds. 

You can use handler. 

I fixed things like this using `debounce` from `rxbindings`