android/camera-samples #9: App crashes while starting CameraFragment

nuhkoca, 2019-05-10:
App crashes at first run.

```
2019-05-10 10:11:08.013 3579-3579/? E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.android.example.cameraxbasic, PID: 3579
    java.lang.ArithmeticException: divide by zero
        at androidx.camera.camera2.impl.SupportedSurfaceCombination.getAllPossibleSizeArrangements(SupportedSurfaceCombination.java:435)
        at androidx.camera.camera2.impl.SupportedSurfaceCombination.getSuggestedResolutions(SupportedSurfaceCombination.java:195)
        at androidx.camera.camera2.impl.Camera2DeviceSurfaceManager.getSuggestedResolutions(Camera2DeviceSurfaceManager.java:185)
        at androidx.camera.core.CameraX.calculateSuggestedResolutions(CameraX.java:449)
        at androidx.camera.core.CameraX.bindToLifecycle(CameraX.java:144)
        at com.android.example.cameraxbasic.fragments.CameraFragment.bindCameraUseCases(CameraFragment.kt:313)
        at com.android.example.cameraxbasic.fragments.CameraFragment.access$bindCameraUseCases(CameraFragment.kt:104)
        at com.android.example.cameraxbasic.fragments.CameraFragment$onViewCreated$1.run(CameraFragment.kt:241)
        at android.os.Handler.handleCallback(Handler.java:878)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:209)
        at android.app.ActivityThread.main(ActivityThread.java:7046)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:486)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:862)
2019-05-10 10:11:32.185 4832-4832/com.android.example.cameraxbasic E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.android.example.cameraxbasic, PID: 4832
    java.lang.ArithmeticException: divide by zero
        at androidx.camera.camera2.impl.SupportedSurfaceCombination.getAllPossibleSizeArrangements(SupportedSurfaceCombination.java:435)
        at androidx.camera.camera2.impl.SupportedSurfaceCombination.getSuggestedResolutions(SupportedSurfaceCombination.java:195)
        at androidx.camera.camera2.impl.Camera2DeviceSurfaceManager.getSuggestedResolutions(Camera2DeviceSurfaceManager.java:185)
        at androidx.camera.core.CameraX.calculateSuggestedResolutions(CameraX.java:449)
        at androidx.camera.core.CameraX.bindToLifecycle(CameraX.java:144)
        at com.android.example.cameraxbasic.fragments.CameraFragment.bindCameraUseCases(CameraFragment.kt:313)
        at com.android.example.cameraxbasic.fragments.CameraFragment.access$bindCameraUseCases(CameraFragment.kt:104)
        at com.android.example.cameraxbasic.fragments.CameraFragment$onViewCreated$1.run(CameraFragment.kt:241)
        at android.os.Handler.handleCallback(Handler.java:878)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:209)
        at android.app.ActivityThread.main(ActivityThread.java:7046)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:486)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:862)

```

tobltobs, 2019-05-14:
Same for me on a Nexus 5 Api 28 emulator.

Snild-Sony, 2019-05-14:
Reported on the CameraX issue tracker: https://issuetracker.google.com/issues/132404864

Go there and star it to get progress updates.

nuhkoca, 2019-05-14:
Hey @Snild-Sony,

Thanks for taking responsibility for the issue. I have done so. Let's get a fast fix, hope!

owahltinez, 2019-05-14:
@Snild-Sony @nuhkoca thanks for flagging this. I can confirm that the team can reproduce the issue and a fix is being worked on. Even though the problem is not related to the sample, I'll keep this issue open until it is fixed in the library.

nuhkoca, 2019-05-16:
@owahltinez thanks for your prompt reply! Waiting for it to be fixed.

owahltinez, 2019-06-06:
This issue has now been fixed, please upgrade to `1.0.0-alpha02` version of CameraX:
```
dependencies {
    ...
    def camerax_version = "1.0.0-alpha02"
    implementation "androidx.camera:camera-core:${camerax_version}"
}
```

closed 2019-06-06 by owahltinez