android/camera-samples #12: About CameraX Preview texture

CaiJingLong, 2019-05-16:
I don't know where CameraX's question should be asked.

I recently tried to develop a flutter plugin with cameraX, but I found that there was no way to simply bind Preview to flutter's Texture.

In the past, I only needed use `camera.setPreviewTexture(surfaceTexture.surfaceTexture())` to bind camera and texture, now I can't find the api.

```kotlin
val preview = Preview(previewConfig)
preview.setOnPreviewOutputUpdateListener {
  // it.surfaceTexture // the surfaceTexture is a readable texture.
}

```

owahltinez, 2019-05-25:
CameraX creates and maintains its own `SurfaceTexture`. I'm not familiar with how Flutter manages its views but you would need to provide Flutter with the `SurfaceTexture` coming from CameraX, not the other way around. Alternatively you can copy frames from one texture to the other but that should be a last resort solution.

CaiJingLong, 2019-05-25:
I also asked this question on Stack Overflow, but came to the conclusion that it was impossible to simply use CameraX, so I decided to give up using CameraX for the time being.
Thanks for answer me.

closed 2019-05-25 by CaiJingLong