android/camera-samples #13: Is sample of CameraX vendor extention jar released?

michaelyang1024, 2019-05-29:


owahltinez, 2019-05-29:
You can take a look at the code sample included in the documentation: https://developer.android.com/training/camerax/vendor-extensions

HanderWei, 2019-05-29:
> You can take a look at the code sample included in the documentation: https://developer.android.com/training/camerax/vendor-extensions

I found code from googlesource.com

https://android.googlesource.com/platform/frameworks/support/+/androidx-master-dev/camera/extensions/src/main/java/androidx/camera/extensions?autodive=0%2F%2F%2F

But I couldn't add `camera-extensins` dependency in build.gradle.

owahltinez, 2019-06-05:
That's correct, the Maven repo does not yet contain artifacts for the vendor extensions. That will probably be part of a future release.

closed 2019-06-05 by owahltinez