android/camera-samples #14: Some device failed on switch lens

Tougee, 2019-05-29:
I've been run this demo in several devices, most of them working great, but failed on Moto X Force(android 7.0). 
The system camera app working well, so I'm not sure it's a bug of cameraX library, and there are also some error log like this:
> E/QCamera2HWI: camera_open failed with return code -87
E/CameraFlashlight: Could not open camera 0: -87
E/CameraClient: initialize: Camera 0: unable to initialize device: Too many users (-87)
E/CameraService: connectHelper: Could not initialize client from HAL module.

owahltinez, 2019-05-30:
Thanks for the feedback. It sounds like the issue may not be with the sample itself, but potentially with the library. Can you please file a bug in the appropriate component of the Android Issue Tracker? It should be https://issuetracker.google.com/components/618491

Tougee, 2019-05-30:
Thanks and filed here: https://issuetracker.google.com/issues/133865294
pixel3XL with latest Q beta version have the same problems

owahltinez, 2019-06-05:
Thanks @Tougee. Closing this one now that we have another bug in the appropriate component.

closed 2019-06-05 by owahltinez

vigneshr87, 2019-07-02:
I have faced same problem and it throws java.lang.IllegalArgumentException: Exceeded max simultaneously bound image capture use cases. exception while switching lenses.  
Just add in CameraFragment -> bindCameraUseCases method
if(preview!=null&& CameraX.isBound(preview))
            CameraX.unbind(preview,imageAnalyzer,imageCapture)
Just unbind usecases that you previously bind and it works

MubasharAlvi, 2021-05-17:
How to switch to front camera on or back camera  CameraX?
CameraSelector.Builder().requireLensFacing(CameraSelector.LENS_FACING_BACK).build()