android/camera-samples #15: Failed to preview when app goes to foreground from background on SC-04E(Samsung Galaxy S4, Android 5.0.1)

Niccari, 2019-05-31:
I've test the CameraFragment in a SC-04E(Samsung Galaxy S4, Android 5.0.1) and the app won't preview when app goes to foreground from background - It said "Surface with size (w=4128, h=2322) and format 0x1 is not valid". In other devices such as a Zenfone 4 max pro and a Google Pixel 2, it works perfectly. 

* SC-04E
(Pressed home button)
> 05-31 14:55:02.813 D/Camera: Use cases [Preview:androidx.camera.core.Preview-edb0fbae-d195-431e-8e5a-5390732e5d8e, ImageCapture:androidx.camera.core.ImageCapture-9c0a037b-355a-48aa-8b18-b8f0b5e9a158, ImageAnalysis:androidx.camera.core.ImageAnalysis-52fc8c63-b185-4577-bbb0-46562c4b1f2b] OFFLINE for camera 0
> 05-31 14:55:02.813 D/Camera: Closing camera: 0
> 05-31 14:55:02.813 V/ActivityThread: updateVisibility : ActivityRecord{20947a6a token=android.os.BinderProxy@222c7695 {com.android.example.cameraxbasic/com.android.example.cameraxbasic.MainActivity}} show : true
> 05-31 14:55:03.354 D/CameraXBasic: Average luminosity: 126.43569661458334. Frames per second: 1.0
> 05-31 14:55:03.404 E/BufferQueueProducer: [unnamed-12479-3] queueBuffer: BufferQueue has been abandoned
> 05-31 14:55:03.404 E/BufferQueueProducer: [unnamed-12479-3] dequeueBuffer: BufferQueue has been abandoned
> 05-31 14:55:03.534 E/BufferQueueProducer: [unnamed-12479-3] queueBuffer: BufferQueue has been abandoned
> 05-31 14:55:03.544 E/BufferQueueProducer: [unnamed-12479-3] dequeueBuffer: BufferQueue has been abandoned
> 05-31 14:55:03.654 E/BufferQueueProducer: [unnamed-12479-3] cancelBuffer: BufferQueue has been abandoned
> 05-31 14:55:03.654 E/BufferQueueProducer: [unnamed-12479-3] cancelBuffer: BufferQueue has been abandoned
> 05-31 14:55:03.664 E/BufferQueueProducer: [unnamed-12479-3] cancelBuffer: BufferQueue has been abandoned
> 05-31 14:55:03.664 E/BufferQueueProducer: [unnamed-12479-3] cancelBuffer: BufferQueue has been abandoned
> 05-31 14:55:03.664 E/BufferQueueProducer: [unnamed-12479-3] cancelBuffer: BufferQueue has been abandoned
> 05-31 14:55:03.674 E/BufferQueueProducer: [unnamed-12479-3] cancelBuffer: BufferQueue has been abandoned
> 05-31 14:55:03.674 E/BufferQueueProducer: [unnamed-12479-3] cancelBuffer: BufferQueue has been abandoned
> 05-31 14:55:03.674 E/BufferQueueProducer: [unnamed-12479-3] cancelBuffer: BufferQueue has been abandoned
> 05-31 14:55:03.674 E/BufferQueueProducer: [unnamed-12479-3] cancelBuffer: BufferQueue has been abandoned
> 05-31 14:55:03.684 E/BufferQueueProducer: [unnamed-12479-3] cancelBuffer: BufferQueue has been abandoned
> 05-31 14:55:03.824 D/Camera: Closing Capture Session
> 05-31 14:55:03.824 D/Camera: CameraDevice.onClosed(): 0
> 05-31 14:55:03.824 D/Camera: Closing Capture Session
> 05-31 14:55:05.246 D/CameraXBasic: Average luminosity: 129.89694986979165. Frames per second: 0.9
> 05-31 14:55:05.866 I/art: Background partial concurrent mark sweep GC freed 232289(5MB) AllocSpace objects, 3(1800KB) LOS objects, 39% free, 22MB/36MB, paused 7.934ms total 104.309ms
> 05-31 14:55:05.876 W/MessageQueue: Handler (android.os.Handler) {34809b5b} sending message to a Handler on a dead thread
>     java.lang.IllegalStateException: Handler (android.os.Handler) {34809b5b} sending message to a Handler on a dead thread
>         at android.os.MessageQueue.enqueueMessage(MessageQueue.java:325)
>         at android.os.Handler.enqueueMessage(Handler.java:631)
>         at android.os.Handler.sendMessageAtFrontOfQueue(Handler.java:623)
>         at android.hardware.camera2.legacy.RequestThreadManager.quit(RequestThreadManager.java:941)
>         at android.hardware.camera2.legacy.LegacyCameraDevice.close(LegacyCameraDevice.java:444)
>         at android.hardware.camera2.legacy.LegacyCameraDevice.finalize(LegacyCameraDevice.java:468)
>         at java.lang.Daemons$FinalizerDaemon.doFinalize(Daemons.java:194)
>         at java.lang.Daemons$FinalizerDaemon.run(Daemons.java:177)
>         at java.lang.Thread.run(Thread.java:818)
> 05-31 14:55:05.876 W/CameraCaptureSession: Session 0: The camera device was already closed: 
>     java.lang.IllegalStateException: CameraDevice was already closed
>         at android.hardware.camera2.impl.CameraDeviceImpl.checkIfCameraClosedOrInError(CameraDeviceImpl.java:1482)
>         at android.hardware.camera2.impl.CameraDeviceImpl.stopRepeating(CameraDeviceImpl.java:677)
>         at android.hardware.camera2.impl.CameraCaptureSessionImpl.close(CameraCaptureSessionImpl.java:328)
>         at android.hardware.camera2.impl.CameraCaptureSessionImpl.finalize(CameraCaptureSessionImpl.java:561)
>         at java.lang.Daemons$FinalizerDaemon.doFinalize(Daemons.java:194)
>         at java.lang.Daemons$FinalizerDaemon.run(Daemons.java:177)
>         at java.lang.Thread.run(Thread.java:818)
> 05-31 14:55:05.876 D/CaptureSession: CameraCaptureSession.onClosed()

(The app has went to foreground)
> 05-31 14:57:07.455 D/CaptureSession: Opening capture session.
> 05-31 14:57:07.465 I/CameraDeviceState: Legacy camera service transitioning to state CONFIGURING
> 05-31 14:57:07.465 I/RequestThread-0: Configure outputs: 3 surfaces configured.
> 05-31 14:57:07.465 D/Camera: app passed NULL surface
> 05-31 14:57:07.465 I/RequestThread-0: configureOutputs - set take picture size to 4128x2322
> 05-31 14:57:07.515 I/CameraDeviceState: Legacy camera service transitioning to state IDLE
> 05-31 14:57:07.515 D/Camera: Use case Preview:androidx.camera.core.Preview-edb0fbae-d195-431e-8e5a-5390732e5d8e RESET for camera 0
> 05-31 14:57:07.525 D/UseCaseAttachState: Active and online use case: [androidx.camera.core.Preview-edb0fbae-d195-431e-8e5a-5390732e5d8e, androidx.camera.core.ImageAnalysis-52fc8c63-b185-4577-bbb0-46562c4b1f2b, androidx.camera.core.ImageCapture-9c0a037b-355a-48aa-8b18-b8f0b5e9a158] for camera: 0
> 05-31 14:57:07.525 D/UseCaseAttachState: All use case: [androidx.camera.core.Preview-edb0fbae-d195-431e-8e5a-5390732e5d8e, androidx.camera.core.ImageAnalysis-52fc8c63-b185-4577-bbb0-46562c4b1f2b, androidx.camera.core.ImageCapture-9c0a037b-355a-48aa-8b18-b8f0b5e9a158] for camera: 0
> 05-31 14:57:07.525 D/Camera: Closing Capture Session
> 05-31 14:57:07.525 D/CaptureSession: Opening capture session.
> 05-31 14:57:07.525 E/CameraDevice-0-LE: Surface with size (w=4128, h=2322) and format 0x1 is not valid, size not in valid set: [1440x1080, 1280x720, 1056x864, 960x720, 720x480, 640x480, 320x240, 176x144]
> 05-31 14:57:07.525 W/CameraDevice-JV-0: Stream configuration failed
> 05-31 14:57:07.525 E/CameraCaptureSession: Session 1: Failed to create capture session; configuration failed
> 05-31 14:57:07.525 D/CaptureSession: CameraCaptureSession.onConfigured()
> 05-31 14:57:07.525 E/CaptureSession: CameraCaptureSession.onConfiguredFailed()
> 05-31 14:57:07.525 D/CaptureSession: CameraCaptureSession.onReady()
> 05-31 14:57:07.525 D/CaptureSession: CameraCaptureSession.onClosed()

* Zenfone 4 Max Pro
[... 33 lines cut ...]
> 2019-05-31 15:12:48.524 D/Camera: CameraDevice is null
> 2019-05-31 15:12:48.524 D/Camera: CameraDevice.onOpened(): 0
> 2019-05-31 15:12:48.525 D/UseCaseAttachState: All use case: [androidx.camera.core.ImageCapture-c311f535-130e-4820-b1ec-bf7edc5e890f, androidx.camera.core.ImageAnalysis-2cad4b2a-0f2c-423b-a9be-da637d372728, androidx.camera.core.Preview-990f8aa3-2403-4cdb-ac42-4592cf850b57] for camera: 0
> 2019-05-31 15:12:48.525 D/Camera: Closing Capture Session
> 2019-05-31 15:12:48.527 D/CaptureSession: Opening capture session.
> 2019-05-31 15:12:48.531 I/CameraDeviceState: Legacy camera service transitioning to state CONFIGURING
> 2019-05-31 15:12:48.531 I/RequestThread-0: Configure outputs: 3 surfaces configured.
> 2019-05-31 15:12:48.531 D/Camera: app passed NULL surface
> 2019-05-31 15:12:48.539 I/RequestThread-0: configureOutputs - set take picture size to 3840x2160
> 2019-05-31 15:12:48.550 I/art: Do partial code cache collection, code=53KB, data=60KB
> 2019-05-31 15:12:48.550 I/art: After code cache collection, code=51KB, data=59KB
> 2019-05-31 15:12:48.550 I/art: Increasing code cache capacity to 256KB
> 2019-05-31 15:12:48.563 I/CameraDeviceState: Legacy camera service transitioning to state IDLE
> 2019-05-31 15:12:48.565 D/CaptureSession: Attempting to send capture request onConfigured
> 2019-05-31 15:12:48.565 D/CaptureSession: Issuing request for session.
> 2019-05-31 15:12:48.568 I/RequestQueue: Repeating capture request set.
> 2019-05-31 15:12:48.568 D/CaptureSession: CameraCaptureSession.onConfigured()
[... cut ...]
> 2019-05-31 15:12:48.574 W/LegacyRequestMapper: Only received metering rectangles with weight 0.
> 2019-05-31 15:12:48.834 I/CameraDeviceState: Legacy camera service transitioning to state CAPTURING
> 2019-05-31 15:12:48.975 D/CameraXBasic: Average luminosity: 42.856079603909464. Frames per second: 1.1
> 2019-05-31 15:12:50.062 D/CameraXBasic: Average luminosity: 81.39527070473251. Frames per second: 29.5
> 2019-05-31 15:12:51.052 D/CameraXBasic: Average luminosity: 92.31874115869341. Frames per second: 35.0

owahltinez, 2019-06-06:
Thanks for the feedback. It sounds like the issue may not be with the sample itself, but potentially with the library. Can you please file a bug in the appropriate component of the Android Issue Tracker? It should be https://issuetracker.google.com/components/618491

Niccari, 2019-06-06:
Thanks for your information. Filed here: https://issuetracker.google.com/issues/134618913.

owahltinez, 2019-06-06:
Thanks @Niccari. Closing this one now that we have another bug in the appropriate component.

closed 2019-06-06 by owahltinez