android/camera-samples #16: Preview is stretched in width

j-ns, 2019-06-03:
While the captured image shows the correct aspect ratio, the preview is obviously stretched in width.  I've tried to set different aspect ratios, e.g. `1/1 `or `viewFinderWidth/viewFinderHeight` to the preview config, as well as transforming the  scale of the viewFinder by `viewFinderMatrix.setScale(outputTextureWidth/viewFinderWidth, outputTextureHeight/viewFinderHeight)`, but none of this could resolve the issue.

(Capture config was setup with the same values as the preview config, i.e. targetAspectRatio and targetResolution)

Phone: Huawei P9 lite
Android: 7.0

owahltinez, 2019-06-06:
Thanks for the feedback. It sounds like the issue may not be with the sample itself, but potentially with the library. Can you please file a bug in the appropriate component of the Android Issue Tracker, including the device information and specific Android build running on the device? The issue tracker should be https://issuetracker.google.com/components/618491

migueledgracia, 2019-07-08:
I had the same issue in the application I was working, when I notice that the sample CameraX app was't suffering the same issue, I notice that they set the `systemUiVisibility` to full screen, so I implement it and that solved the camera preview being stretch

```
    private fun setImmersiveMode() {
        // body is root view in my layout
        body?.systemUiVisibility =
                View.SYSTEM_UI_FLAG_FULLSCREEN or
                View.SYSTEM_UI_FLAG_LAYOUT_FULLSCREEN
    }
```

Satheeshkummari, 2019-07-18:
I too had the same issue in Camera 2 API. I resolved it by configuring preview frame resolution that occupy full device height and width.
it is fixed for all the devices which has frame Aspect Ratio matches to device AR.
And for the devices which doesn't have frames matches to device AR, issue still exists.
but rather ugly stretch i configured matched frame to device and showing softnav bar and allowing to small amount of stretch.
I have suffered a week to fix this issue.

saschpe, 2019-08-02:
https://issuetracker.google.com/issues/138814839

owahltinez, 2019-08-04:
Thanks! Closing this issue in favor of the one in AOSP

closed 2019-08-04 by owahltinez

VuManhHung97, 2019-10-07:
> Tôi gặp vấn đề tương tự trong ứng dụng tôi đang làm việc, khi tôi nhận thấy ứng dụng CameraX mẫu không gặp vấn đề tương tự, tôi nhận thấy rằng họ đặt chế độ `systemUiVisibility`toàn màn hình, vì vậy tôi đã triển khai nó và giải quyết việc xem trước camera bị kéo dài
> 
> ```
>     private fun setImmersiveMode() {
>         // body is root view in my layout
>         body?.systemUiVisibility =
>                 View.SYSTEM_UI_FLAG_FULLSCREEN or
>                 View.SYSTEM_UI_FLAG_LAYOUT_FULLSCREEN
>     }
> ```
It does not work for me