android/camera-samples #17: Access denied finding property "vendor.camera.aux.packagelist"

ajayVinay, 2019-06-05:


owahltinez, 2019-06-06:
Can you please provide a few more details? Sample code that triggers your issue?

ajayVinay, 2019-06-06:
I did this .....

/////#######################////
package com.example.cameraxapp

import android.Manifest
import android.content.pm.PackageManager
import android.graphics.Matrix
import android.os.Bundle
import android.os.Handler
import android.os.HandlerThread
import android.util.Log
import android.util.Rational
import android.util.Size
import android.view.Surface
import android.view.TextureView
import android.view.ViewGroup
import android.widget.ImageButton
import android.widget.Toast
import androidx.appcompat.app.AppCompatActivity
import androidx.camera.core.*
import androidx.core.content.ContextCompat
import androidx.lifecycle.LifecycleOwner
import java.io.File
import java.nio.ByteBuffer
import java.util.concurrent.TimeUnit

// This is an arbitrary number we are using to keep tab of the permission
// request. Where an app has multiple context for requesting permission,
// this can help differentiate the different contexts
private const val REQUEST_CODE_PERMISSIONS = 10

// This is an array of all the permission specified in the manifest
private val REQUIRED_PERMISSIONS = arrayOf(Manifest.permission.CAMERA)

class MainActivity : AppCompatActivity(),LifecycleOwner {

    private lateinit var viewFinder: TextureView

    override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)
        setContentView(R.layout.activity_main)

    }
    // this  method create for implement cameraX operations
    // create configuration object for the viewfinder use case
    private fun startCamera() {

        val previewConfig = PreviewConfig.Builder().apply {

            setTargetAspectRatio(Rational(1, 1))
            setTargetResolution(Size(640, 640))

        }.build()

        // Build the viewFinder use case
        val preview = Preview(previewConfig)

        preview.setOnPreviewOutputUpdateListener {

            // To update the surfaceTexture, we have to remove it and re-add it
            val parent = viewFinder.parent as ViewGroup
            parent.removeView(viewFinder)
            parent.addView(viewFinder, 0)

            // it :- this is return current value
            viewFinder.surfaceTexture = it.surfaceTexture
            updateTransform()
        }

[... 132 lines cut ...]
        get(data)   // Copy the buffer into a byte array
        return data // Return the byte array
    }

    override fun analyze(image: ImageProxy, rotationDegrees: Int) {
        val currentTimestamp = System.currentTimeMillis()
        // Calculate the average luma no more often than every second
        if (currentTimestamp - lastAnalyzedTimestamp >=
            TimeUnit.SECONDS.toMillis(1)) {
            // Since format in ImageAnalysis is YUV, image.planes[0]
            // contains the Y (luminance) plane
            val buffer = image.planes[0].buffer
            // Extract image data from callback object
            val data = buffer.toByteArray()
            // Convert the data into an array of pixel values
            val pixels = data.map { it.toInt() and 0xFF }
            // Compute average luminance for the image
            val luma = pixels.average()
            // Log the new luma value
            Log.d("CameraXApp", "Average luminosity: $luma")
            // Update timestamp of last analyzed frame
            lastAnalyzedTimestamp = currentTimestamp
        }
    }
}

owahltinez, 2019-06-07:
A few follow-up questions:
1. What specific line is throwing the error?
2. Can you paste the stack trace as well?
3. Does this happen when you run the sample as-is?
4. If you modified the sample, can you tell us what you changed that causes the error?

ajayVinay, 2019-06-07:
1.There is no specific line throwing error
3.Yes ,this happen when am i rum sample code 
4. No modification sample code

owahltinez, 2019-06-11:
Thanks for the feedback. It sounds like the issue may not be with the sample itself, but potentially with the library. Can you please file a bug in the appropriate component of the Android Issue Tracker? It should be https://issuetracker.google.com/components/618491. Please include information about the specific device that you are seeing this on.

owahltinez, 2019-07-18:
Closing after 30+ days of inactivity. Feel free to reopen if you are still encountering this issue.

closed 2019-07-18 by owahltinez

RajappaR, 2019-11-15:
This issue not resolved yet, i am also facing the same issue

P1514, 2019-11-21:
Hello just in case someone like me is having this issue, I'm using a Nokia 6.1 only managed to have this code to work with the following camera2 version **'androidx.camera:camera-camera2:1.0.0-alpha03**'
firebase-ml didn't seem to impact here.