android/camera-samples #18: Device Gallery Displays Deleted Photos

Enshaedn, 2019-06-05:
First time adding an issue to a GitHub project. I'd like to start with saying that I really appreciate the demo app - it helped me a lot, and I'm excited for CameraX. 

I did notice in my own app, and then tested with this demo app, that missing image preview icons were left in the phone's gallery after deleting an image from the app photo gallery. I would have to separately go back through and delete the thumbnails in the phone gallery after deleting the files. 

I found some help via [StackOverflow](https://stackoverflow.com/questions/10716642/android-deleting-an-image) solutions, but I'm not sure if it is the best approach. Currently I've added: 

context.sendBroadcast(Intent(Intent.ACTION_MEDIA_SCANNER_SCAN_FILE, Uri.fromFile(mediaFile)))

just after the mediaFile.delete() line, and that seems to have solved the problem. Another answer suggested using a ContentResolver to delete the image, but again, being new to programming I'm not sure which is the "best" option.

owahltinez, 2019-06-07:
Hi @Enshaedn, congrats on your first issue and thank you for the feedback!

Can you give us a little more information about the environment you are seeing this in? What device are you seeing this in? What is the gallery app, and which version?

I just tested the sample app on a Pixel 3 device using Google Photos as the gallery app, and images deleted from the sample app get immediately removed from Google Photos.

Enshaedn, 2019-06-07:
Thanks @owahltinez. :)

I noticed it first on my Samsung Galaxy S10 in the app called "Gallery". Presumably a Samsung app as it comes on the phone and is updated from the Galaxy Store. The version is 10.0.02.108.

I can also see it happen on a Pixel 2 emulator running API 28. However there I see it in just the file system itself. 

If I go to the specific folder (Android>media>com.android.example.cameraxbasic>CameraX Basic) I can see correctly that I have a single image taken today. 

[image]

If I just go to Files, open the Nav Drawer, and click Images > CameraX Basic, then I see all the previously deleted files still. I can click on them, but nothing loads because they are deleted. For whatever reason that this time most of them are still even showing thumbnails of what they were except for the single all white one. 

[image]

Hopefully that is helpful information.

owahltinez, 2019-06-11:
Thank you, that's very helpful! My guess of what's happening is that some gallery apps are caching images, rather than querying the file system every time. I think that adding something similar to `context.sendBroadcast(Intent(Intent.ACTION_MEDIA_SCANNER_SCAN_FILE, Uri.fromFile(mediaFile)))` is a fair solution. Let us investigate a bit more, we can keep this bug open for now.

commit 6801d9b2 (2019-06-28): Implement safe args and delete ImageUtils

- Fixes #18
- Fixes #19
- Fixes #26
- Implement safe args
- Crop circles using Glide
- Delete unneeded ImageUtils

Change-Id: I928aef500771c0ba9122203a2295cd2f141e3e41
closed 2019-06-28 by owahltinez