android/camera-samples #20: Not taking sensor orientation into account

tsuharesu, 2019-06-06:
It seems that the sensor orientation is not being taken into account, ie my photos are sideways. 

I remember other samples to use something to rotate the image correctly https://github.com/googlesamples/android-Camera2Basic/blob/master/kotlinApp/Application/src/main/java/com/example/android/camera2basic/Camera2BasicFragment.kt

Is this a problem with the sample or the library?

owahltinez, 2019-06-07:
The library handles sensor orientation. Internally, it is done via `CameraInfo.getSensorRotationDegrees()`. You can see the relevant interface [here](https://android.googlesource.com/platform/frameworks/support/+/androidx-master-dev/camera/core/src/main/java/androidx/camera/core/CameraInfo.java#54).

What device are you using that takes photos sideways? Is the preview sideways as well?

tsuharesu, 2019-06-07:
I'm using a Huawei P30 Pro.
Checking the image in the sample app shows correctly, but in my Files app, or Solid Explorer, even when copying to my Mac, it shows sideways. 
Maybe those apps are not handling the orientation tag correctly? (SE shows Orientation as 7)? 

Would be nice to save the photo in the correct orientation, that's what the other app does, or not?

leninexe, 2019-07-01:
I'm having the same problem on Pixel. Apparently there's no way to store the rotation EXIF?

owahltinez, 2019-07-02:
The issue could be related to how you are saving your file, it's possible that somewhere the EXIF information is stripped. I have tried taking multiple photos with different orientations on a Pixel 3 XL phone but they all appear "upright" when viewing them with the Files app as well as after sharing them with other apps (like Gmail).

> Would be nice to save the photo in the correct orientation, that's what the other app does, or not?

To my understanding, the other app is doing the same -- setting the EXIF metadata but not actually rotating the image, but I could be wrong about that.

tsuharesu, 2019-07-02:
The EXIF info is being saved correctly, but that doesn't mean that other apps will handle the EXIF correctly, right? If I'm using a server to save the image and open it on another platform (i.e. iOS) or in the browser, the photo can be sideways or something.

What I'm doing in my app is that after saving the image, I rotate it. The gist: https://gist.github.com/tsuharesu/833dc98230855f32e0f9719d60ca5e1e

```kotlin
private suspend fun rotateImageCorrectly() = withContext(Dispatchers.IO) {
        val sourceBitmap = MediaStore.Images.Media.getBitmap(contentResolver, photoFile.toUri())

        val exif = ExifInterface(photoFile.inputStream())
        val rotation = exif.getAttributeInt(ExifInterface.TAG_ORIENTATION, ExifInterface.ORIENTATION_NORMAL)
        val rotationInDegrees = when (rotation) {
            ExifInterface.ORIENTATION_ROTATE_90 -> 90
            ExifInterface.ORIENTATION_ROTATE_180 -> 180
            ExifInterface.ORIENTATION_ROTATE_270 -> 270
            ExifInterface.ORIENTATION_TRANSVERSE -> -90
            ExifInterface.ORIENTATION_TRANSPOSE -> -270
            else -> 0
        }
        val matrix = Matrix().apply {
            if (rotation != 0) preRotate(rotationInDegrees.toFloat())
        }

        val rotatedBitmap =
            Bitmap.createBitmap(sourceBitmap, 0, 0, sourceBitmap.width, sourceBitmap.height, matrix, true)

        rotatedBitmap.compress(Bitmap.CompressFormat.JPEG, 100, FileOutputStream(photoFile))

        sourceBitmap.recycle()
        rotatedBitmap.recycle()
    }
```

owahltinez, 2019-07-02:
> The EXIF info is being saved correctly, but that doesn't mean that other apps will handle the EXIF correctly, right?

That's correct, but we can't feasibly accommodate for bugs in other applications. EXIF metadata embedded in JPEG images has been a [well-established spec](https://www.loc.gov/preservation/digital/formats/fdd/fdd000147.shtml) for a long time now. 

By doing what is being suggested with your code snippet, you are re-compressing the JPEG image and thus there is a potential loss of quality (however insignificant that might be). You will also be compressing the image using Bitmap's compression method, whereas the JPEG provided to you by the CameraX API will be coming from the camera module itself, which is encoding the image in a way that has likely been optimized for the hardware.

tsuharesu, 2019-07-02:
@owahltinez you are correct with all your observations. I just wanted to share how I did to save the photo in the correct orientation.

owahltinez, 2019-07-18:
Closing since we now understand that this is not an issue with the sample itself, just lack of proper EXIF support in other photo viewing software. Feel free to reopen if I'm not understanding correctly.

closed 2019-07-18 by owahltinez

adminnu, 2019-10-12:
Good afternoon!

The problem really exists and for a long time I could not find a worthy solution. When using the front camera, the image on different devices is sideways or upside down, I'm not talking about the fact that it is a mirror.
In jpg itself there are meta tags and when manually specifying .setTargetAspectRatio it rotates.

This problem existed even before the advent of camerax, but we are well aware that camerax was implemented to simplify the implementation of camera solutions on all devices. But as the problem was, it remained, if you type in the search engine a request, why the image from the front camera is upside down and how to fix it, there will be many requests.

Here is my code:
```
void startCamera() {
        CameraX.unbindAll();

        String typeCamera = getIntent().getStringExtra("typeCamera");
        CameraX.LensFacing LensFacing = CameraX.LensFacing.BACK;
        int Rotation = getWindowManager().getDefaultDisplay().getRotation();

        if (typeCamera.equals("front")) {
            try {
                if (CameraX.hasCameraWithLensFacing(CameraX.LensFacing.FRONT)) {
                    LensFacing = CameraX.LensFacing.FRONT;
                    //Rotation = Surface.ROTATION_180;
                } else {
                    LensFacing = CameraX.LensFacing.BACK;
                }
            } catch (CameraInfoUnavailableException e) {
                e.printStackTrace();
            }
        }

        int aspRatioW = cameraView.getWidth(); //get width of screen
        int aspRatioH = cameraView.getHeight(); //get height
        Rational asp = new Rational(aspRatioW, aspRatioH); //aspect ratio
        Size screen = new Size(aspRatioW, aspRatioH); //size of the screen

        PreviewConfig pConfig = new PreviewConfig.Builder()
                .setTargetAspectRatio(asp)
                .setTargetResolution(screen)
                .setLensFacing(LensFacing)
                .build();
        Preview preview = new Preview(pConfig); //lets build it

        preview.setOnPreviewOutputUpdateListener(
                new Preview.OnPreviewOutputUpdateListener() {
                    //to update the surface texture we have to destroy it first, then re-add it
                    @Override
                    public void onUpdated(Preview.PreviewOutput output) {
                        ViewGroup parent = (ViewGroup) cameraView.getParent();
                        parent.removeView(cameraView);
                        parent.addView(cameraView, 0);

                        cameraView.setSurfaceTexture(output.getSurfaceTexture());
                        updateTransform();
                    }
                });

        /* image capture */
        ImageCaptureConfig imgCapConfig = new ImageCaptureConfig.Builder()
                .setCaptureMode(ImageCapture.CaptureMode.MIN_LATENCY)
                .setTargetRotation(Rotation)
                .setLensFacing(LensFacing)
                .build();
        final ImageCapture imgCap = new ImageCapture(imgCapConfig);

        final ImageButton cameraButton = findViewById(R.id.camera_button);

        cameraButton.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View v) {
                Log.v(TAG, "CameraButton clicked");
                cameraButton.setVisibility(View.INVISIBLE);
[... 50 lines cut ...]
                });
            }
        });

        /* image analyser */

        ImageAnalysisConfig imgAConfig = new ImageAnalysisConfig.Builder()
                .setImageReaderMode(ImageAnalysis.ImageReaderMode.ACQUIRE_LATEST_IMAGE)
                .setLensFacing(LensFacing)
                .build();

        ImageAnalysis analysis = new ImageAnalysis(imgAConfig);

        analysis.setAnalyzer(
                new ImageAnalysis.Analyzer() {
                    @Override
                    public void analyze(ImageProxy image, int rotationDegrees) {
                        //y'all can add code to analyse stuff here idek go wild.
                    }
                });

        //bind to lifecycle:
        CameraX.bindToLifecycle(this, analysis, imgCap, preview);
    }
```

mankind-evolve, 2021-12-20:
Thanks very much !