android/camera-samples #21: App crashes after display orientation change

dnhyde, 2019-06-08:
The app crashes when I change the phone orientation from portrait to landscape.

Phone: Xiaomi Redmi 5 Plus
OS: Android 8.1.0, API27

```2019-06-08 19:03:14.348 10259-10332/com.android.example.cameraxbasic D/CaptureSession: Opening capture session.
2019-06-08 19:03:14.353 10259-10332/com.android.example.cameraxbasic I/CameraDeviceState: Legacy camera service transitioning to state CONFIGURING
2019-06-08 19:03:14.354 10259-10540/com.android.example.cameraxbasic I/RequestThread-0: Configure outputs: 3 surfaces configured.
2019-06-08 19:03:14.354 10259-10540/com.android.example.cameraxbasic D/Camera: app passed NULL surface
2019-06-08 19:03:14.354 10259-10540/com.android.example.cameraxbasic E/BufferQueueProducer: [ImageReader-4000x2000f100m2-10259-2] connect: already connected (cur=4 req=4)
2019-06-08 19:03:14.355 10259-10540/com.android.example.cameraxbasic E/Legacy-CameraDevice-JNI: connectSurface: Unable to connect to surface, error Invalid argument (-22).
2019-06-08 19:03:14.355 10259-10540/com.android.example.cameraxbasic E/Legacy-CameraDevice-JNI: LegacyCameraDevice_nativeConnectSurface: Error while configuring surface Invalid argument (-22).

    --------- beginning of crash
2019-06-08 19:03:14.356 10259-10540/com.android.example.cameraxbasic E/AndroidRuntime: FATAL EXCEPTION: RequestThread-0
    Process: com.android.example.cameraxbasic, PID: 10259
    java.lang.UnsupportedOperationException: Unknown error -22
        at android.hardware.camera2.legacy.LegacyExceptionUtils.throwOnError(LegacyExceptionUtils.java:77)
        at android.hardware.camera2.legacy.LegacyCameraDevice.connectSurface(LegacyCameraDevice.java:694)
        at android.hardware.camera2.legacy.RequestThreadManager.configureOutputs(RequestThreadManager.java:406)
        at android.hardware.camera2.legacy.RequestThreadManager.-wrap0(Unknown Source:0)
        at android.hardware.camera2.legacy.RequestThreadManager$5.handleMessage(RequestThreadManager.java:713)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:176)
        at android.os.HandlerThread.run(HandlerThread.java:65)
2019-06-08 19:03:14.633 10259-10540/com.android.example.cameraxbasic I/Process: Sending signal. PID: 10259 SIG: 9
```

I can post the full verbose stacktrace if necessary.

I must say that device manufacturer does not support camera2 (`CameraCharacteristics.INFO_SUPPORTED_HARDWARE_LEVEL_LEGACY`) and it's quite a hassle to work with this device...

owahltinez, 2019-06-11:
Thanks for the feedback. It sounds like the issue may not be with the sample itself, but potentially with the library. Can you please file a bug in the appropriate component of the Android Issue Tracker? It should be https://issuetracker.google.com/components/618491

dnhyde, 2019-06-21:
Thanks, I reported it in the issue tracker!

owahltinez, 2019-07-02:
Thank you @dnhyde, closing this issue.

closed 2019-07-02 by owahltinez