android/camera-samples #22: Photo look darker than normal camera

C2oWisComing, 2019-06-09:
Photo from camerax example look darker than from normal camera

Device : Samsung Galaxy Tab S2 (SM-T819Y)
OS : Android 7.0

Normal Camera Application
[image]

CameraX Example Application
[image]

owahltinez, 2019-06-11:
Thanks for the feedback. It sounds like the issue may not be with the sample itself, but potentially with the library, since the sample just uses the auto-brightness provided by CameraX. Can you please file a bug in the appropriate component of the Android Issue Tracker? It should be https://issuetracker.google.com/components/618491

owahltinez, 2019-07-18:
Closing after 30+ days of inactivity.

closed 2019-07-18 by owahltinez

sarthakssg5, 2020-01-02:
> Thanks for the feedback. It sounds like the issue may not be with the sample itself, but potentially with the library, since the sample just uses the auto-brightness provided by CameraX. Can you [...]

I don't think so. This issue was present in the older repo as well but was closed due to the migration of the repo. 
[https://github.com/googlearchive/android-Camera2Basic/issues/53](url)

thebino, 2020-01-21:
I observed the same behaviour on CameraX. Looks like its an library issue.
Bug report: https://issuetracker.google.com/issues/148045396