android/camera-samples #23: How to enable bokehImageCapture extension ?

AthiraRoslin, 2019-06-11:
In the developer site, enabling vendor extensions are elaborated. But when I try to use the same code, its not showing any extensions at all. Can anybody help on this ?

AthiraRoslin, 2019-06-11:
This is the code I'm refering to,

import androidx.camera.extensions.BokehExtender

fun onCreate() {
    // Create a Builder same as in normal workflow.
    val builder = ImageCaptureConfig.Builder()

    // Create a Extender object which can be used to apply extension
    // configurations.
    val bokehImageCapture = BokehImageCaptureExtender.create(builder)

    // Query if extension is available (optional).
    if (bokehImageCapture.isExtensionAvailable()) {
        // Enable the extension if available.
        bokehImageCapture.enableExtension()
    }

    // Finish constructing configuration with the same flow as when not using
    // extensions.
    val config = builder.build()
    val useCase = ImageCapture(config)
    CameraX.bindToLifecycle(this as LifecycleOwner, useCase)
}

When I try the same, it is not importing the extensions.

owahltinez, 2019-06-27:
@AthiraRoslin what device are you observing this behavior on?

owahltinez, 2019-07-18:
Closing since extensions are not currently implemented in our sample. Feel free to open an issue in the AOSP issue tracker or continue the conversation in our [developer community forums](g.co/camerax/developers).

closed 2019-07-18 by owahltinez