android/camera-samples #25: Provide example of how to use takePicture(OnImageCapturedListener)?

chunhunghan, 2019-06-12:
Could you provide a demo about how to capture and handle in-memory buffer image?

SakuyaCN, 2019-06-28:
```
val buffer = image!!.planes[0].buffer
                val bytes = ByteArray(buffer.capacity()).also {
                    buffer[it]
                }
```

chunhunghan, 2019-06-28:
My question is a little miss leading, I have update question to use Intent without File Uri.
Thanks

owahltinez, 2019-07-02:
Hi @chunhunghan 

This is an issue with the sample, we should implement receiving the [MediaStore.ACTION_IMAGE_CAPTURE](https://developer.android.com/reference/android/provider/MediaStore#ACTION_IMAGE_CAPTURE) intent.

To implement the [MediaStore.ACTION_IMAGE_CAPTURE](https://developer.android.com/reference/android/provider/MediaStore#ACTION_IMAGE_CAPTURE) intent, you have to take care of two cases:
1. The caller passed an [EXTRA_OUTPUT](https://developer.android.com/reference/android/provider/MediaStore.html#EXTRA_OUTPUT) with a URI where the file must be saved.
2. The caller did not pass an [EXTRA_OUTPUT](https://developer.android.com/reference/android/provider/MediaStore.html#EXTRA_OUTPUT) so the Activity must return a Bitmap as part of the result data.

The result must be returned using the [setResult](https://developer.android.com/reference/android/app/Activity.html#setResult(int,%20android.content.Intent)) method of `MainActivity` before calling `finish()`.

Please let us know if you run into any issues. In a future update, we will implement this as part of the sample.

closed 2019-07-03 by chunhunghan