android/camera-samples #26: Cancel coroutineContext

BulatMukhutdinov, 2019-06-12:
The CameraFragment implements CoroutineScope so it have to implement CoroutineContext. For that context you create new instance of Job and hold in class filed. Later, onDestroy() you call ```job.cancel()```. However,  it is better to cancel the coroutineContext itself. So that you also don't need to hold explicit reference to the job.
Moreover you could use 
```class CameraFragment : Fragment(), CoroutineScope by CoroutineScope(Dispatchers.Default)``` so that you can remove couroutineContext and it's explicit cancellation in onDestroy(). 
You can take a look to the best practices described here https://kotlinlang.org/docs/reference/coroutines/coroutine-context-and-dispatchers.html#coroutine-scope

owahltinez, 2019-06-27:
Thanks for the feedback. Admittedly, I am far from being an expert on Kotlin coroutines.

Implementing CoroutineScope is a temporary solution until AndroidX Lifecycle's runtime adds the relevant utilities to Activities and Fragments, which is already available as of version `2.2.0-alpha01` in `androidx.lifecycle:lifecycle-runtime-ktx`. Then, inside of a Fragment we can do:

```
lifecycleScope.launch(coroutineContext) {
    // async code here...
}
```

Take a look at [the docs](https://developer.android.com/topic/libraries/architecture/coroutines) for more information, and feel free to provide more feedback if you don't think that would be an appropriate solution to this issue.

BulatMukhutdinov, 2019-06-27:
Thank you for sharing that. Yes, it would be just right solution

commit 6801d9b2 (2019-06-28): Implement safe args and delete ImageUtils

- Fixes #18
- Fixes #19
- Fixes #26
- Implement safe args
- Crop circles using Glide
- Delete unneeded ImageUtils

Change-Id: I928aef500771c0ba9122203a2295cd2f141e3e41
closed 2019-06-27 by BulatMukhutdinov