android/camera-samples #27: Capturing DNG

vork, 2019-06-20:
Hi,

so in theory you can change the [ImageFormat](https://developer.android.com/reference/androidx/camera/core/UseCase.html#setImageFormat(int)) in the UseCase to something like [RAW_SENSOR](https://developer.android.com/reference/android/graphics/ImageFormat.html#RAW_SENSOR). However, each [Callback of ImageCapture](https://developer.android.com/reference/androidx/camera/core/ImageCapture.OnImageCapturedListener.html) mentions that

> The image capture method fully supports the JPEG format.

So I have not looked at what the API does behind the scenes, but is this a bug in the documentation or is this really not possible because of behind the scenes plumbing currently?

owahltinez, 2019-07-02:
Hi @vork 

In theory it *may* work but it's untested and you are treading the waters of device-dependent implementations -- since not all devices support RAW format. The focus of CameraX so far has been on JPEG because all devices must support it to pass [CTS](https://source.android.com/compatibility/cts/camera-hal).

owahltinez, 2019-07-18:
Closing since this is not a sample issue. Feel free to continue the discussion in our [developer community forums](g.co/camerax/developers).

closed 2019-07-18 by owahltinez