android/camera-samples #30: App crashes after switching to front camera

m-o, 2019-06-28:
Hi, I installed the app and when I try to switch to front camera the app crashes.
LG G7, Andorid 9

> 2019-06-28 13:55:47.745 729-849/com.android.example.cameraxbasic E/AndroidRuntime: FATAL EXCEPTION: CameraX-

    Process: com.android.example.cameraxbasic, PID: 729
    java.lang.IllegalArgumentException: CaptureRequest contains unconfigured Input/Output Surface!
        at android.hardware.camera2.CaptureRequest.convertSurfaceToStreamId(CaptureRequest.java:674)
        at android.hardware.camera2.impl.CameraDeviceImpl.submitCaptureRequest(CameraDeviceImpl.java:1093)
        at android.hardware.camera2.impl.CameraDeviceImpl.setRepeatingRequest(CameraDeviceImpl.java:1139)
        at android.hardware.camera2.impl.CameraCaptureSessionImpl.setRepeatingRequest(CameraCaptureSessionImpl.java:298)
        at androidx.camera.camera2.impl.CaptureSession.issueRepeatingCaptureRequests(CaptureSession.java:443)
        at androidx.camera.camera2.impl.CaptureSession$StateCallback.onConfigured(CaptureSession.java:604)
        at androidx.camera.core.CameraCaptureSessionStateCallbacks$ComboSessionStateCallback.onConfigured(CameraCaptureSessionStateCallbacks.java:115)
        at android.hardware.camera2.impl.CallbackProxies$SessionStateCallbackProxy.lambda$onConfigured$0(CallbackProxies.java:53)
        at android.hardware.camera2.impl.-$$Lambda$CallbackProxies$SessionStateCallbackProxy$soW0qC12Osypoky6AfL3P2-TeDw.run(Unknown Source:4)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)<

owahltinez, 2019-07-02:
Thanks for the feedback. It sounds like the issue may not be with the sample itself, but potentially with the library. Can you please file a bug in the appropriate component of the Android Issue Tracker and reply here once the you have opened the issue there? It should be under https://issuetracker.google.com/components/618491

owahltinez, 2019-07-30:
Closing due to inactivity

closed 2019-07-30 by owahltinez

titanwalking, 2019-08-04:
Hi, I have an LG G7 ThinQ too, and this issue occurs time to time while switching to front cam.