android/camera-samples #31: Question: how to enable focus on touch in cameraX

OleksandrGrument, 2019-07-02:
Hello, can you please explain how to enable focus on touch in camerax library. Is it feature currently available. There is no information about it in usage example and in official documentation. Thanks

owahltinez, 2019-07-02:
Hi @OleksandrGrument 

The tap-to-focus API is currently under development. If you have feedback or have any further questions, you can get in touch with the CameraX team in our developer forum at [g.co/camerax/developers](http://g.co/camerax/developers)

migueledgracia, 2019-07-08:
I did manage to implement a simple focus on touch, that will focus in the `ImageView` that display a square, this won't focus where you touch in the screen, but it might give you an idea how to implement it using `TouchListener`

```
    protected fun onTouchFocus() {
        preview?.focus(
                Rect(iv_frame.left, iv_frame.top, iv_frame.right, iv_frame.bottom),
                Rect(iv_frame.left, iv_frame.top, iv_frame.right, iv_frame.bottom))
    }
```

owahltinez, 2019-07-18:
Thanks for the code snippet @migueledgracia. Hopefully we will implement this feature in our sample in the future.

closed 2019-07-18 by owahltinez