android/camera-samples #33: ImageAnalysis always returns 640x480 images

zorouyang, 2019-07-08:
ImageAnalysis setTargetResolution(Size(1280, 720)), but ImageProxy always return 640x480 image.

migueledgracia, 2019-07-08:
Base on what the documentation saids:

>         /**
         * Sets the resolution of the intended target from this configuration.
         *
         * <p>The target resolution attempts to establish a minimum bound for the image resolution.
         * The actual image resolution will be the closest available resolution in size that is not
         * smaller than the target resolution, as determined by the Camera implementation. However,
         * if no resolution exists that is equal to or larger than the target resolution, the
         * nearest available resolution smaller than the target resolution will be chosen.
         *
         * @param resolution The target resolution to choose from supported output sizes list.
         * @return The current Builder.
         */

Base on what I understood, If the camera of the device doesn't support that resolution, the nearest available resolution smaller or grater than the target resolution will be chosen.

Probably the device you'r using might not support that specific resolution and is falling back to the nearest one that it support, which might be 640x480

zorouyang, 2019-07-09:
> Base on what the documentation saids:
> 
> > 
> 
> ```
>     /** 
>      * Sets the resolution of the intended target from this configuration.
>      *
>      * <p>The target resolution attempts to establish a minimum bound for the image resolution.
>      * The actual image resolution will be the closest available resolution in size that is not
>      * smaller than the target resolution, as determined by the Camera implementation. However,
>      * if no resolution exists that is equal to or larger than the target resolution, the
>      * nearest available resolution smaller than the target resolution will be chosen.
>      *
>      * @param resolution The target resolution to choose from supported output sizes list.
>      * @return The current Builder.
>      */
> ```
> 
> Base on what I understood, If the camera of the device doesn't support that resolution, the nearest available resolution smaller or grater than the target resolution will be chosen.
> 
> Probably the device you'r using might not support that specific resolution and is falling back to the nearest one that it support, which might be 640x480
> 
> You might wanna target a much higher resolution
> `setTargetResolution(Size(4256, 2832))`
> That did the trick for me

Hi, 
Thank you for your reply, which solved my problem.

owahltinez, 2019-07-30:
Closing as solved, thanks @migueledgracia !

closed 2019-07-30 by owahltinez