android/camera-samples #35: Black TextureSurface after navigation

Sparki2002, 2019-07-12:
I use a `BottomNavigationView` to navigate between my camera fragment and another one.
On the start of my app the camera preview is loaded correctly.
If I navigate to my second fragment and back again, `CameraX` is successfully bound to the lifecycle, but the `TextureSurface` is black.

I can resolve this by adding `Thread.sleep(330)` before `CameraX.bindToLifecycle()` but this is a bad solution, because it adds some delay before showing the preview.

What can I do to solve this problem without this delay?

owahltinez, 2019-07-18:
Can you please provide some code that lets us reproduce this issue?

Tooto, 2019-07-31:
I think, its happen when u rebind camera to lifecycle.
I have black screen os some devices (ex. Alcatel 5059D) when trying to switch between front and back cameras. 
I have error in logcat:
` E/Camera: CameraDevice.onError(): 1 with error: ERROR_MAX_CAMERAS_IN_USE`

owahltinez, 2019-09-04:
I think this issue should be at least partially alleviated with `alpha04`. Please reopen if it still reproduces, and provide specific device and code to reproduce this issue consistently.

closed 2019-09-04 by owahltinez