android/camera-samples #37: Bug at CameraXBasic example : camera switch

davinnovation, 2019-07-15:
When calling bindCameraUseCases at `camera_switch_button` click listener, it doesn't work correctly with Galaxy S10+, Note 8

If I add `CameraX.unbindAll()` code to `bindCameraUseCases()` function at start line, it works normarlly. 

```
private fun bindCameraUseCases() {
        CameraX.unbindAll()
        ...
}
```

Tooto, 2019-07-16:
Work for me! (Samsung note 9)

owahltinez, 2019-07-18:
@davinnovation does this happen 100% of the time? Do you see any errors in logcat?

Tooto, 2019-07-18:
> @davinnovation does this happen 100% of the time? Do you see any errors in logcat?

100%. Tested on sgs6, sgnote 9.

`
2019-07-18 13:08:40.998 373-373/com.motify.client E/CameraFragment$updateCameraUi: java.lang.IllegalArgumentException: Exceeded max simultaneously bound image capture use cases.
        at androidx.camera.camera2.impl.UseCaseSurfaceOccupancyManager.checkUseCaseLimitNotExceeded(UseCaseSurfaceOccupancyManager.java:61)
        at androidx.camera.camera2.impl.Camera2DeviceSurfaceManager.getSuggestedResolutions(Camera2DeviceSurfaceManager.java:150)
        at androidx.camera.core.CameraX.calculateSuggestedResolutions(CameraX.java:494)
        at androidx.camera.core.CameraX.bindToLifecycle(CameraX.java:188)
`

davinnovation, 2019-07-18:
@owahltinez Yes. When I press the switch button, 100%. - Galaxy S10+ confirmed

In logcat, error occurred just one time with this message
```
2019-07-18 20:24:29.535 4188-4239/com.android.example.cameraxbasic E/Camera: CameraDevice.onError(): 1 with error: ERROR_MAX_CAMERAS_IN_USE
```
After this, error message doesn't shown and only this message has logged when
```
2019-07-18 20:23:13.619 392-392/com.android.example.cameraxbasic D/ViewRootImpl@705f68f[MainActivity]: ViewPostIme pointer 0
2019-07-18 20:23:13.644 392-392/com.android.example.cameraxbasic D/ViewRootImpl@705f68f[MainActivity]: ViewPostIme pointer 1
```

owahltinez, 2019-07-23:
OK, the problem is that when the camera switch button is pressed, we are trying to bind all camera use cases without unbinding the previous ones first. Thanks for flagging!

I think that the proper solution here would be to call `CameraX.unbindAll()` inside of the switch button callback, rather than inside of the `bindCameraUseCases()` function.

commit 73c7a063 (2019-07-25): Update CameraXBasic sample to alpha03

- Update to CameraX alpha03
- Update to Gradle plugin 3.4.2
- Add sample table to repo README
- Fix #36 (switching to front camera causes crash)
- Fix #37 (camera switch button not working)
- Fix #40 (revoking permissions causes crash)
- Fix b/137221008 (gradlew missing execute permission)

Change-Id: Id78f6eb0aff88440c19b2d74760308f8f914a413
closed 2019-07-25 by owahltinez