android/camera-samples #38: IllegalArgumentException: No supported surface combination is found

Tooto, 2019-07-17:
I added VideoCapture to the sample

```
// Create a configuration object for the video use case
        val videoCaptureConfig = VideoCaptureConfig.Builder().apply {
            setLensFacing(lensFacing)
            // We request aspect ratio but no resolution to match preview config but letting
            // CameraX optimize for whatever specific resolution best fits requested capture mode
            setTargetAspectRatio(screenAspectRatio)
            // Set initial target rotation, we will have to call this again if rotation changes
            // during the lifecycle of this use case
            setTargetRotation(viewFinder.display.rotation)
        }.build()

        videoCapture = VideoCapture(videoCaptureConfig)
```
And modify this line: 
`  CameraX.bindToLifecycle(viewLifecycleOwner, preview, imageCapture, videoCapture)`

On some devices y catch exception:
```
    java.lang.IllegalArgumentException: No supported surface combination is found for camera device - Id : 0.  May be attempting to bind too many use cases.
        at androidx.camera.camera2.impl.Camera2DeviceSurfaceManager.getSuggestedResolutions(Camera2DeviceSurfaceManager.java:176)
```
I cant find any way to fix it. Can you help me?

owahltinez, 2019-07-18:
The issue here is that you are trying to add too many use cases to the same camera session. To better understand the limitations of how many different streams you can get out of the camera, [take a look at the documentation](https://developer.android.com/reference/android/hardware/camera2/CameraDevice.html?source=post_page---------------------------#createCaptureSession(android.hardware.camera2.params.SessionConfiguration)) or [read our blog post about this topic](https://medium.com/androiddevelopers/using-multiple-camera-streams-simultaneously-bf9488a29482).

owahltinez, 2019-07-30:
Closing this issue. If you have any more questions about use cases not currently implemented in the sample (like `VideoCapture`) feel free to reach out at the [developer community forums](g.co/camerax/developers).

closed 2019-07-30 by owahltinez

mainakbiswas, 2020-05-20:
I have faced the same issue today. What I wanted to achieve is this. I want to record a video and simultaneously analyse each frame of the video for some image analysis. How should I proceed?

`private fun startCamera() {
        // This is the Texture View where the camera will be rendered
        val viewFinder = binding.viewFinder

        // The ratio for the output video and preview
        val ratio = AspectRatio.RATIO_16_9

        // The Configuration of how we want to preview the camera
        val previewConfig = PreviewConfig.Builder().apply {
            setTargetAspectRatio(ratio) // setting the aspect ration
            setLensFacing(lensFacing) // setting the lens facing (front or back)
            setTargetRotation(viewFinder.display.rotation)// setting the rotation of the camera
        }.build()

        // Create an instance of Camera Preview
        preview = AutoFitPreviewBuilder.build(previewConfig, viewFinder)

        // The Configuration of how we want to capture the video
        val videoCaptureConfig = VideoCaptureConfig.Builder().apply {
            setTargetAspectRatio(ratio) // setting the aspect ration
            setLensFacing(lensFacing) // setting the lens facing (front or back)
            setVideoFrameRate(30) // setting the frame rate to 24 fps
            setTargetRotation(viewFinder.display.rotation) // setting the rotation of the camera
        }.build()

        videoCapture = VideoCapture(videoCaptureConfig)

        binding.fabRecordVideo.setOnClickListener { recordVideo(videoCapture) }

        // Add all the use cases to the CameraX
        val imageAnalysisConfig = ImageAnalysisConfig.Builder().apply {
            setImageReaderMode(ImageAnalysis.ImageReaderMode.ACQUIRE_NEXT_IMAGE)
        }.build()
        val imageAnalysis = ImageAnalysis(imageAnalysisConfig).apply {
            val analyzer = ImageAnalysis.Analyzer { image, rotationDegrees ->
                val buffer = image.planes[0].buffer
                buffer.rewind()
                val bmp = Bitmap.createBitmap(image.width, image.height, Bitmap.Config.ARGB_8888).also {
                    it.copyPixelsFromBuffer(buffer)
                }
            }
            setAnalyzer(requireContext().mainExecutor(), analyzer)
        }
        CameraX.bindToLifecycle(viewLifecycleOwner, preview, videoCapture, imageAnalysis)`

jaydeepw, 2021-05-26:
> I have faced the same issue today. What I wanted to achieve is this. I want to record a video and simultaneously analyse each frame of the video for some image analysis. How should I proceed? [...]

Does this solution work. I am doing something similar and I am still getting the exception

zhambylgaziz, 2022-03-02:
> Does this solution work. I am doing something similar and I am still getting the exception

Hello,
I have the same issue. Can you tell me, have you managed the problem?

dreamchaser96, 2022-03-21:
> Hello, I have the same issue. Can you tell me, have you managed the problem?

have you managed to figure this out? Same issue

jplipata, 2022-12-05:
I'm having the same issue on Pixel 7, however my code works on Samsung S21. The [proposed solutions in this thread](https://github.com/android/camera-samples/issues/38#issuecomment-512670590) are based on Camera2. Hoping that by now there's a solution using CameraX