android/camera-samples #40: Crash CameraXBasic example when revoking CAMERA permission

astamato, 2019-07-22:
Hey there! 

So I've installed the app example `CameraXBasic`, have it running, granted camera permission. Then I navigate to settings, revoke that permission, go back to the app and I got a crash. I was expecting rather a warning or a dialog of some kind.

This is not the behaviour in example `Camera2BasicKotlin` where this exact flow is managed gracefully, and you see a dialog "This sample needs camera permission." 

Device: Android Pixel Q, patch 05/07.

Logcat:

```
     --------- beginning of crash
2019-07-22 17:14:46.154 19934-21147/? E/AndroidRuntime: FATAL EXCEPTION: CameraX-
    Process: com.android.example.cameraxbasic, PID: 19934
    java.lang.SecurityException: validateClientPermissionsLocked:976: Caller "com.android.example.cameraxbasic" (PID 10402, UID 19934) cannot open camera "0" without camera permission
        at android.hardware.camera2.CameraManager.throwAsPublicException(CameraManager.java:793)
        at android.hardware.camera2.CameraManager.openCameraDeviceUserAsync(CameraManager.java:426)
        at android.hardware.camera2.CameraManager.openCameraForUid(CameraManager.java:588)
        at android.hardware.camera2.CameraManager.openCamera(CameraManager.java:516)
        at androidx.camera.camera2.impl.Camera.openCameraDevice(Camera.java:595)
        at androidx.camera.camera2.impl.Camera.open(Camera.java:151)
        at androidx.camera.camera2.impl.Camera.addOnlineUseCase(Camera.java:505)
        at androidx.camera.camera2.impl.Camera$10.run(Camera.java:488)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:214)
        at android.os.HandlerThread.run(HandlerThread.java:67)
     Caused by: android.os.ServiceSpecificException: validateClientPermissionsLocked:976: Caller "com.android.example.cameraxbasic" (PID 10402, UID 19934) cannot open camera "0" without camera permission (code 1)
        at android.os.Parcel.createException(Parcel.java:2085)
        at android.os.Parcel.readException(Parcel.java:2039)
        at android.os.Parcel.readException(Parcel.java:1987)
        at android.hardware.ICameraService$Stub$Proxy.connectDevice(ICameraService.java:583)
        at android.hardware.camera2.CameraManager.openCameraDeviceUserAsync(CameraManager.java:389)
        at android.hardware.camera2.CameraManager.openCameraForUid(CameraManager.java:588) 
        at android.hardware.camera2.CameraManager.openCamera(CameraManager.java:516) 
        at androidx.camera.camera2.impl.Camera.openCameraDevice(Camera.java:595) 
        at androidx.camera.camera2.impl.Camera.open(Camera.java:151) 
        at androidx.camera.camera2.impl.Camera.addOnlineUseCase(Camera.java:505) 
        at androidx.camera.camera2.impl.Camera$10.run(Camera.java:488) 
        at android.os.Handler.handleCallback(Handler.java:883) 
        at android.os.Handler.dispatchMessage(Handler.java:100) 
        at android.os.Looper.loop(Looper.java:214) 
        at android.os.HandlerThread.run(HandlerThread.java:67) 
```

Does this make sense? Any suggestion as to how to deal with this scenario with cameraX?

Cheers! (:

owahltinez, 2019-07-23:
Does this happen if you kill the app and restart it after denying the camera permission? I think the problem might be that the app is only being paused, then the permission is revoked and we are not currently checking for permissions inside of the resume event.

astamato, 2019-07-23:
You don't need to kill the app necessarily, just navigate to settings, revoke permission and then back to app.

`CameraFragment` will be recreated (you see the calls of the whole flow onCreate, onCreateView, onViewCreated and so on) but without permissions it blows up. `CameraFragment` should be responsible of verifying it has appropriate permissions before trying to call CameraX, and maybe not delegating this to `PermissionsFragment`.

Just thought it would be worth mentioning there is this crash since it's very easy to reproduce.

commit 73c7a063 (2019-07-25): Update CameraXBasic sample to alpha03

- Update to CameraX alpha03
- Update to Gradle plugin 3.4.2
- Add sample table to repo README
- Fix #36 (switching to front camera causes crash)
- Fix #37 (camera switch button not working)
- Fix #40 (revoking permissions causes crash)
- Fix b/137221008 (gradlew missing execute permission)

Change-Id: Id78f6eb0aff88440c19b2d74760308f8f914a413
closed 2019-07-25 by owahltinez