android/camera-samples #41: Using ImageAnalysis is extremely slow on low-spec Nokia 5.1

bilsou, 2019-07-23:
The `CameraXBasic` sample runs really slow on Nokia 5.1 because of the `imageAnalyzer` when binding the use in `CameraX.bindToLifecycle(viewLifecycleOwner, preview, imageAnalyzer)`. If one removes the `imageAnalyzer` parameter, the frame rate gets normal again. 

Are you doing in copy before calling the base function `fun analyze(image: ImageProxy, rotationDegrees: Int) {` ?

jxdom, 2019-07-25:
Me too,my device is meizhu5.1

owahltinez, 2019-09-04:
@bilsou there is no image copy as far as I'm aware. How are you configuring the image analysis use case? Are you using the [ACQUIRE_LATEST_IMAGE](https://developer.android.com/reference/androidx/camera/core/ImageAnalysis.ImageReaderMode.html#ACQUIRE_LATEST_IMAGE) mode?

owahltinez, 2019-10-10:
Closing for now, please reopen if you are still experiencing this issue when using ACQUIRE_LATEST_IMAGE mode

closed 2019-10-10 by owahltinez

mohammad-ali-dehlvi, 2022-09-23:
I am facing the same issue with ACQUIRE_LATEST_IMAGE mode

adeeb1201, 2023-05-22:
I am also facing the issue in Emulator, Pixel 5 API 29. Will this happen in realtime too?