android/camera-samples #42: Is there any way to handle camera error?

Tooto, 2019-07-26:
I've got 
`CameraDevice.onError(): 0 with error: ERROR_CAMERA_DEVICE` in logcat.

Is there any way to intercept this error in camera fragment?

Stacktrace:
`
  android.hardware.camera2.CameraAccessException: CAMERA_ERROR (3): The camera device has encountered a serious error
        at android.hardware.camera2.impl.CameraDeviceImpl.checkIfCameraClosedOrInError(CameraDeviceImpl.java:2417)
        at android.hardware.camera2.impl.CameraDeviceImpl.stopRepeating(CameraDeviceImpl.java:1106)
        at android.hardware.camera2.impl.CameraCaptureSessionImpl.close(CameraCaptureSessionImpl.java:527)
        at androidx.camera.camera2.impl.CaptureSession.release(CaptureSession.java:357)
        at androidx.camera.camera2.impl.Camera.resetCaptureSession(Camera.java:692)
        at androidx.camera.camera2.impl.Camera$StateCallback.onError(Camera.java:976)
        at androidx.camera.core.CameraDeviceStateCallbacks$ComboDeviceStateCallback.onError(CameraDeviceStateCallbacks.java:120)
        at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.notifyError(CameraDeviceImpl.java:1910)
        at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.lambda$Sm85frAzwGZVMAK-NE_gwckYXVQ(Unknown Source:0)
        at android.hardware.camera2.impl.-$$Lambda$CameraDeviceImpl$CameraDeviceCallbacks$Sm85frAzwGZVMAK-NE_gwckYXVQ.accept(Unknown Source:8)
        at com.android.internal.util.function.pooled.PooledLambdaImpl.doInvoke(PooledLambdaImpl.java:258)
        at com.android.internal.util.function.pooled.PooledLambdaImpl.invoke(PooledLambdaImpl.java:182)
        at com.android.internal.util.function.pooled.OmniFunction.run(OmniFunction.java:77)
        at android.os.Handler.handleCallback(Handler.java:891)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:207)
        at android.os.HandlerThread.run(HandlerThread.java:65)
`

owahltinez, 2019-07-30:
Unfortunately, there's no good way to catch these errors currently. How is this error triggered in your app?

Tooto, 2019-07-30:
This problem happens on Huawei/Honor devices when i want to record video with .setMaxResolution

```
  // Create a configuration object for the video use case
        val videoCaptureConfig = VideoCaptureConfig.Builder().apply {
            setLensFacing(lensFacing)
            // We request aspect ratio but no resolution to match preview config but letting
            // CameraX optimize for whatever specific resolution best fits requested capture mode
            setTargetAspectRatio(screenAspectRatio)
            if (!presenterMyProgress.isDeviceHuawei()) { // да, нарушение декларации  об архитектуре
                setMaxResolution(preferVideoSize)
            }
            // Set initial target rotation, we will have to call this again if rotation changes
            // during the lifecycle of this use case
            setTargetRotation(viewFinder.display.rotation)
        }.build()
```

owahltinez, 2019-09-04:
Thanks for reporting this, I think this may be an issue with the library! Can you confirm if it still happens in `alpha04`?

owahltinez, 2019-10-02:
@Tooto any luck trying to reproduce with the latest version? We just upgraded the sample to `alpha05`.

owahltinez, 2019-12-23:
Closing due to inactivity, feel free to reopen if you are still hitting this issue.

closed 2019-12-23 by owahltinez

AbhisKmr, 2020-05-14:
I'm facing this same crash please reopen this issue

darkplataform, 2020-09-06:
I am facing the same issue as well, HUAWEI P30 @Tooto  your suggestion was good to avoid the issue, however I would like to be able to set default or max resolution.