android/camera-samples #43: IllegalArgumentException: Suggested resolution map missing resolution for camera 1

Reginer, 2019-07-29:
setLensFacing(CameraX.LensFacing.FRONT)  Before initialization

owahltinez, 2019-07-30:
Hi @Reginer can you please provide a few more details here? Is this an issue you are running into with the sample, or are you modifying the sample in order to trigger this issue?

Reginer, 2019-07-30:
[This File](https://github.com/android/camera/blob/master/CameraXBasic/app/src/main/java/com/android/example/cameraxbasic/fragments/CameraFragment.kt)
line 270   setLensFacing(CameraX.LensFacing.FRONT)

Reginer, 2019-07-30:
[image]

owahltinez, 2019-08-03:
Sorry, I'm still unable to reproduce the issue. What device are you seeing this in?

Reginer, 2019-08-03:
Maybe because it's an MTK system.
It workd on Huawei phones.

owahltinez, 2019-08-03:
It sounds like a device specific issue. Can you please file a bug at https://issuetracker.google.com/components/618491 and provide the specific device and model you are using?

Reginer, 2019-08-05:
sorry,Demo not worked in Huawei device。
This is my code:
[image]
[image]

Error is:
```
Caused by: java.lang.IllegalArgumentException: Suggested resolution map missing resolution for camera 1
        at androidx.camera.core.Preview.onSuggestedResolutionUpdated(Preview.java:433)
        at androidx.camera.core.UseCase.updateSuggestedResolution(UseCase.java:380)
        at androidx.camera.core.CameraX.calculateSuggestedResolutions(CameraX.java:503)
        at androidx.camera.core.CameraX.bindToLifecycle(CameraX.java:188)
```

MingLeo, 2019-08-17:
I'm facing the same issue too. Works perfectly fine when I set "setLensFacing(CameraX.LensFacing.BACK)". Issue only occurs when "setLensFacing(CameraX.LensFacing.FRONT)", i think it is beacause it is unable to perform resolutionMap to CameraId=1 even though CameraManager is able to detect the presence of CameraId=1 (FRONT) through getCameraCharacteristics().

[image]

[image]

[image]

I am testing on my phone which is a Samsung S7, running on Android version 8.0.0 which I believe to be an Android OREO (API 26) version.

MingLeo, 2019-08-17:
Sorry if i have any misconception, from my limited understanding of CameraX, looking into the revolutionary & lifesaving bindToLifecycle() method of CameraX.java class, it takes in and groups the usecases like Preview, imageCapture, imageAnalysis which are pass in as parameters, 

[image]
[image]
[image]

under the CameraX.java file, there is an implementation  "for (String cameraId : newCameraIdUseCaseMap.keySet()){....}" which is suppose to get the suggested resolutions and update the use case session configuration as can be seen in its code implementation 
[image]

& Under the PreviewConfig.java class, which implements interface CameraDeviceConfig, there is a method specified "CameraX.LensFacing getLensFacing" to retrieve the direction of the Lens facing
[image]

Hence from my limited amount of understanding, it seems like CameraX is suppose to help me handle passing CameraId=1 and help me perform/handle resolution map by passing CameraId =1 (FRONT )into all the respective usecases so that they are able to detect the target resolution for CameraId =1 (FRONT), and all i have to do is to specify setLensFacing(direction) under the previewConfig which happens to be a usecase for CameraX.bindToLifecycle() method.

I understand that in Camera2 api we would have to handle/match our own Camera Instance as below, but now seems to be managed by the new CameraX api.
[image]

MingLeo, 2019-08-17:
My sincere apologies for the long post, I'm an undergraduate student hailing from sunny island of SINGAPORE flying the flag alone where the coding community culture is almost absent(unless you count expensive heavily-incentivized codecamps as one), new to android & currently working on a feature that requires the use of the front-facing camera & I'm stuck at this juncture. I have searched up stack overflow and there has not been any useful solutions, asking questions on stack overflow has personally been a toxic environment for a beginner like me and I have always wish to reach out to someone pro in the community. Sincere appreciation if you read this and reply. 

There are tons of Online paid tutorials that just teach basic functions and are painfully outdated as compared to the face pace of change push out by Google. Sometimes i feel like i just mastered a new feature and then a new update comes along, previous things become deprecated, no longer supported and I become super lost in exasperation, having no idea where to start from, the online documentation is not updated and people are unable to provide any solutions since these features are so new, alpha, beta or newly realeased RC, and I have to go through the pain of binge watching the google I/O videos just to find out what's new, what's updated, find out if there are any googe codelabs tutorial examples(which sometimes dont even work. Gawd...... one hell of an experience from me).  Sometimes I just wish i have contacts in google so that I can be one call away from them. It seems like they only provide support for corporates who are working along side them to test any new features and report bugs back, but the rest of us is just drowning in the deep ocean.  Sincerely apologize for the long post.

owahltinez, 2019-09-04:
@MingLeo I highly encourage you to join our [developer forums](g.co/camerax/developers), you can rant all you want there 🙂

Regarding your specific issue, can you please confirm it is happening with `alpha04`? There are some changes happening around the internal logic that picks a resolution (related to [this](https://issuetracker.google.com/issues/140194349)), but that probably won't be released until `alpha05`.

owahltinez, 2019-10-02:
We just upgraded the sample to use `alpha05`. Feel free to reopen if this issue is still reproducible.

closed 2019-10-02 by owahltinez