android/camera-samples #44: Crash in the Camera2RawFragment on a Samsung J5

EarlOfEgo, 2019-07-31:
The Camera2Raw sample is crashing on a Samsung J5 8.1.0 device.
See the stacktrace:

```
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.example.android.camera2raw, PID: 14934
    java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.Object android.hardware.camera2.CameraCharacteristics.get(android.hardware.camera2.CameraCharacteristics$Key)' on a null object reference
        at com.example.android.camera2raw.Camera2RawFragment.configureTransform(Camera2RawFragment.java:1046)
        at com.example.android.camera2raw.Camera2RawFragment.access$000(Camera2RawFragment.java:121)
        at com.example.android.camera2raw.Camera2RawFragment$1.onSurfaceTextureAvailable(Camera2RawFragment.java:212)
        at android.view.TextureView.getHardwareLayer(TextureView.java:390)
        at android.view.TextureView.draw(TextureView.java:339)
        at android.view.View.updateDisplayListIfDirty(View.java:19428)
        at android.view.View.draw(View.java:20206)
        at android.view.ViewGroup.drawChild(ViewGroup.java:4459)
        at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4245)
        at android.view.View.updateDisplayListIfDirty(View.java:19419)
        at android.view.View.draw(View.java:20206)
        at android.view.ViewGroup.drawChild(ViewGroup.java:4459)
        at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4245)
        at android.view.View.draw(View.java:20486)
        at android.view.View.updateDisplayListIfDirty(View.java:19428)
        at android.view.View.draw(View.java:20206)
        at android.view.ViewGroup.drawChild(ViewGroup.java:4459)
        at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4245)
        at android.view.View.updateDisplayListIfDirty(View.java:19419)
        at android.view.View.draw(View.java:20206)
        at android.view.ViewGroup.drawChild(ViewGroup.java:4459)
        at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4245)
        at android.view.View.updateDisplayListIfDirty(View.java:19419)
        at android.view.View.draw(View.java:20206)
        at android.view.ViewGroup.drawChild(ViewGroup.java:4459)
        at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4245)
        at android.view.View.draw(View.java:20486)
        at com.android.internal.policy.DecorView.draw(DecorView.java:990)
        at android.view.View.updateDisplayListIfDirty(View.java:19428)
        at android.view.ThreadedRenderer.updateViewTreeDisplayList(ThreadedRenderer.java:719)
        at android.view.ThreadedRenderer.updateRootDisplayList(ThreadedRenderer.java:725)
        at android.view.ThreadedRenderer.draw(ThreadedRenderer.java:833)
        at android.view.ViewRootImpl.draw(ViewRootImpl.java:3567)
        at android.view.ViewRootImpl.performDraw(ViewRootImpl.java:3354)
        at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2889)
        at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1843)
        at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:7978)
        at android.view.Choreographer$CallbackRecord.run(Choreographer.java:911)
        at android.view.Choreographer.doCallbacks(Choreographer.java:723)
        at android.view.Choreographer.doFrame(Choreographer.java:658)
        at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:897)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:7000)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:441)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1408)
Application terminated.

```

owahltinez, 2019-08-04:
Do you know if the Samsung J5 8.1.0 supports raw camera capture? The crash indicates that the `mCharacteristics` variable is never set, which could happen if the following condition is met for all the available cameras:

```kotlin
if (!contains(characteristics.get(
    CameraCharacteristics.REQUEST_AVAILABLE_CAPABILITIES),
    CameraCharacteristics.REQUEST_AVAILABLE_CAPABILITIES_RAW)) {
    continue;
}
```

EarlOfEgo, 2019-08-05:
@owahltinez yes you are right, the Samsung J5 does not support raw image capturing.

closed 2019-09-04 by owahltinez