android/camera-samples #45: CameraX: Camera Crashes After a Few Rotations

breakingthebuild, 2019-08-01:
Running the sample app located here: https://github.com/android/camera/tree/master/CameraXBasic on my Pixel3.

After rotating the app 4 or 5 times, the TextureView goes black and the live camera feed is gone. Killing the app and restarting it, doesn't fix the issue. Attached are the logs. I also had the same issue with the sample app for Camera2. Not sure if this is an issue with the sample apps or an underlying issue with the camera x library (version: 1.0.0-alpha03).

owahltinez, 2019-09-04:
Hey @breakingthebuild (cool username!), can you confirm if this still happens in `alpha04`? I can't see any logs attached to your previous comment.

breakingthebuild, 2019-09-04:
So I have just come to realize that my camera is not working. So even with the native camera app, it will work the first time but then I have to restart my phone in order for it work. Sometime that doesn't even work. We can close this ticket. Sorry for wasting your time.

closed 2019-09-04 by breakingthebuild