android/camera-samples #47: Can't find cameraId.

yangwuan55, 2019-08-02:
This is the crash log.
```
2019-08-02 22:13:15.877 30884-30884/com.android.example.cameraxbasic E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.android.example.cameraxbasic, PID: 30884
    java.lang.IllegalArgumentException: Fail to find supported surface info - CameraId:null
        at androidx.camera.camera2.impl.Camera2DeviceSurfaceManager.requiresCorrectedAspectRatio(Camera2DeviceSurfaceManager.java:273)
        at androidx.camera.core.Preview.updateUseCaseConfig(Preview.java:387)
        at androidx.camera.core.UseCase.<init>(UseCase.java:92)
        at androidx.camera.core.Preview.<init>(Preview.java:99)
        at com.android.example.cameraxbasic.utils.AutoFitPreviewBuilder.<init>(AutoFitPreviewBuilder.kt:91)
        at com.android.example.cameraxbasic.utils.AutoFitPreviewBuilder.<init>(AutoFitPreviewBuilder.kt:39)
        at com.android.example.cameraxbasic.utils.AutoFitPreviewBuilder$Companion.build(AutoFitPreviewBuilder.kt:243)
        at com.android.example.cameraxbasic.fragments.CameraFragment.bindCameraUseCases(CameraFragment.kt:279)
        at com.android.example.cameraxbasic.fragments.CameraFragment.access$bindCameraUseCases(CameraFragment.kt:89)
        at com.android.example.cameraxbasic.fragments.CameraFragment$onViewCreated$1.run(CameraFragment.kt:249)
        at android.os.Handler.handleCallback(Handler.java:755)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6141)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:912)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:802)

```

If I don't set the ``` lensFacing ```
```

2019-08-02 22:08:40.401 17580-17580/? E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.android.example.cameraxbasic, PID: 17580
    java.lang.IllegalArgumentException: Unable to get camera ID for use case androidx.camera.core.Preview-a61300f3-0284-41a8-8845-92d868cda597
        at androidx.camera.camera2.impl.Camera2DeviceSurfaceManager.getCameraIdFromConfig(Camera2DeviceSurfaceManager.java:310)
        at androidx.camera.camera2.impl.Camera2DeviceSurfaceManager.requiresCorrectedAspectRatio(Camera2DeviceSurfaceManager.java:268)
        at androidx.camera.core.Preview.updateUseCaseConfig(Preview.java:387)
        at androidx.camera.core.UseCase.<init>(UseCase.java:92)
        at androidx.camera.core.Preview.<init>(Preview.java:99)
        at com.android.example.cameraxbasic.utils.AutoFitPreviewBuilder.<init>(AutoFitPreviewBuilder.kt:91)
        at com.android.example.cameraxbasic.utils.AutoFitPreviewBuilder.<init>(AutoFitPreviewBuilder.kt:39)
        at com.android.example.cameraxbasic.utils.AutoFitPreviewBuilder$Companion.build(AutoFitPreviewBuilder.kt:243)
        at com.android.example.cameraxbasic.fragments.CameraFragment.bindCameraUseCases(CameraFragment.kt:279)
        at com.android.example.cameraxbasic.fragments.CameraFragment.access$bindCameraUseCases(CameraFragment.kt:89)
        at com.android.example.cameraxbasic.fragments.CameraFragment$onViewCreated$1.run(CameraFragment.kt:249)
        at android.os.Handler.handleCallback(Handler.java:755)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6141)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:912)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:802)
     Caused by: java.lang.IllegalArgumentException: Option does not exist: Option{id=camerax.core.camera.lensFacing, valueClass=class androidx.camera.core.CameraX$LensFacing, token=null}
        at androidx.camera.core.OptionsBundle.retrieveOption(OptionsBundle.java:107)
        at androidx.camera.core.PreviewConfig.retrieveOption(PreviewConfig.java:73)
        at androidx.camera.core.PreviewConfig.getLensFacing(PreviewConfig.java:178)
        at androidx.camera.camera2.impl.Camera2DeviceSurfaceManager.getCameraIdFromConfig(Camera2DeviceSurfaceManager.java:307)
        at androidx.camera.camera2.impl.Camera2DeviceSurfaceManager.requiresCorrectedAspectRatio(Camera2DeviceSurfaceManager.java:268) 
        at androidx.camera.core.Preview.updateUseCaseConfig(Preview.java:387) 
        at androidx.camera.core.UseCase.<init>(UseCase.java:92) 
        at androidx.camera.core.Preview.<init>(Preview.java:99) 
        at com.android.example.cameraxbasic.utils.AutoFitPreviewBuilder.<init>(AutoFitPreviewBuilder.kt:91) 
        at com.android.example.cameraxbasic.utils.AutoFitPreviewBuilder.<init>(AutoFitPreviewBuilder.kt:39) 
        at com.android.example.cameraxbasic.utils.AutoFitPreviewBuilder$Companion.build(AutoFitPreviewBuilder.kt:243) 
        at com.android.example.cameraxbasic.fragments.CameraFragment.bindCameraUseCases(CameraFragment.kt:279) 
        at com.android.example.cameraxbasic.fragments.CameraFragment.access$bindCameraUseCases(CameraFragment.kt:89) 
        at com.android.example.cameraxbasic.fragments.CameraFragment$onViewCreated$1.run(CameraFragment.kt:249) 
        at android.os.Handler.handleCallback(Handler.java:755) 
        at android.os.Handler.dispatchMessage(Handler.java:95) 
        at android.os.Looper.loop(Looper.java:154) 
        at android.app.ActivityThread.main(ActivityThread.java:6141) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:912) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:802) 

```

I use this demo it work well,so what config is wrong?
https://github.com/pranaypatel512/CameraXDemo

yangwuan55, 2019-08-03:
At this demo https://github.com/pranaypatel512/CameraXDemo I change the version to 1.0.0.alpha03 has the same error.So I think this is the bug of alpha03.
But the CameraXBasic demo still can't run at alpha01.

owahltinez, 2019-08-03:
USB cameras are not supported by CameraX at this time ☹️

closed 2019-08-03 by owahltinez

yangwuan55, 2019-08-04:
@owahltinez But the alpha01 can work well.I think this bug is not at usb,because the camera app is work.

owahltinez, 2019-08-04:
My theory is that there must be a bug in alpha01 that prevents filtering out external cameras. I think the team would love to learn what you are trying to do with more details, can you please send a message to [g.co/camerax/developers](g.co/camerax/developers) explaining how you are trying to use external cameras with CameraX? You can also just reply to this thread which discusses support of USB cameras: https://groups.google.com/a/android.com/forum/#!topic/camerax-developers/9cLEOpfvux8

yangwuan55, 2019-08-05:
I did not use the android thing,it's just a normal android system on arm.
The system camera app is work well,so the camera is a normal standard for the android system.
I will use the camera2 api,because the report bug process is too complicated.
Any how,thanks.

owahltinez, 2019-08-05:
USB cameras are supported by the Android framework starting in Android P, that applies to all flavours of Android including phones, things or even TV. However, USB cameras are not supported by the CameraX library. If you find the bug report process complicated, feel free to chime into the discussion at [g.co/camerax/developers](g.co/camerax/developers), we would appreciate your input!

VuManhHung97, 2019-11-08:
I also encountered this case on nokia 6.0 and sum sung j7 prime

manav-mygate, 2020-01-02:
same issue in YOGA Tab 3 8”. Is there any fix available?

latinosamuel, 2020-05-09:
Can anyone tell if the USB cameras are already supported by CameraX, using the latest version?

yangwuan55, 2020-05-13:
I think the problem is not the USB Camera,but there is the wrong place to commit issues for CameraX.

> Can anyone tell if the USB cameras are already supported by CameraX, using the latest version?

arpitcybernetyx, 2020-09-22:
@owahltinez I am also facing the same issue in our company panel running with Android8 . However the same is working in our another device with another android image running android7 . So ,  can the issue be with android AOSP code rather than android version .

AnyLifeZLB, 2023-05-22:
I use the  latest version camerax, emm USB cameras are not supported still . 2023-05-22