android/camera-samples #50: CameraX: NullPointerException when trying to record video

Twometer, 2019-08-09:
I have a simple video recording app set up like this:
```java
VideoCaptureConfig videoCaptureConfig = new VideoCaptureConfig.Builder()
        .setLensFacing(CameraX.LensFacing.BACK)
        .setTargetAspectRatio(new Rational(16, 9))
        .setTargetRotation(viewFinder.getDisplay().getRotation())
        .build();
videoCapture = new VideoCapture(videoCaptureConfig);

CameraX.bindToLifecycle(this, preview, videoCapture);
```
...
```java
 videoCapture.startRecording(file, new VideoCapture.OnVideoSavedListener() { ... });
```

which leads to the following NullPointerException:
```
    java.lang.NullPointerException: Attempt to invoke virtual method 'void android.media.AudioRecord.startRecording()' on a null object reference
        at androidx.camera.core.VideoCapture.startRecording(VideoCapture.java:262)
        at androidx.camera.core.VideoCapture.startRecording(VideoCapture.java:236)
        at de.twometer.stream.cam.CameraActivity.lambda$onCreate$1$CameraActivity(CameraActivity.java:77)
        at de.twometer.stream.cam.-$$Lambda$CameraActivity$MvEAnqH_74mMgbMAcSbB_pNoJ3k.onClick(Unknown Source:4)
        at android.view.View.performClick(View.java:7352)
        at android.view.View.performClickInternal(View.java:7318)
        at android.view.View.access$3200(View.java:846)
        at android.view.View$PerformClick.run(View.java:27807)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7037)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:494)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:965)
```

I am using CameraX version `1.0.0-alpha04`

Twometer, 2019-08-09:
Nevermind, I am just stupid and forgot to request audio permission.

closed 2019-08-09 by Twometer

Link184, 2020-09-22:
The issue still persist event with audio permission. CameraX version `1.0.0-beta09`

jaydeepw, 2021-05-26:
I have `RECORD_AUDIO` permission however, I am still getting this exception when trying to capture the video.

ssjsaha, 2021-07-06:
Hi there , Did you guys had the solution? Im facing the same problem here !

chief10x, 2022-05-26:
I still face the same issue

temcguir, 2022-05-26:
`androidx.camera.core.VideoCapture` is not a supported public API, and should give lint errors when you try to use it as it is marked as restricted.

`androidx.camera.video.VideoCapture`, which is part of the `camera-video` artifact is the officially supported use case for recording video. Are you running into this problem with that class?

namdhis, 2022-09-23:
> Hi there , Did you guys had the solution? Im facing the same problem here !

Did you find the way to record video by camera X without using audio permision yet?

temcguir, 2022-09-23:
`androidx.camera.video.VideoCapture` should not require the audio permission unless you call [`PendingRecording.withAudioEnabled()`](https://developer.android.com/reference/androidx/camera/video/PendingRecording#withAudioEnabled()). 

Please see [this guide](https://developer.android.com/training/camerax/video-capture#using-videocapture-api) on how to use the new VideoCapture API.