android/camera-samples #51: Crash when returning back to camera app from background

shardul, 2019-08-13:
When resuming the app from background app is crashing with surface error.

Steps to replicate

1. Start the sample app
2. Press Home Button
3. Start the app again from recent apps

OS, Android 9.0
Device: Google Pixel 2 XL

Stacktrace:
Process: com.android.example.cameraxbasic, PID: 3428
    java.lang.IllegalArgumentException: Cannot setSurfaceTexture to a released SurfaceTexture
        at android.view.TextureView.setSurfaceTexture(TextureView.java:744)
        at com.android.example.cameraxbasic.utils.AutoFitPreviewBuilder$1.onUpdated(AutoFitPreviewBuilder.kt:106)
        at androidx.camera.core.Preview.updateOutput(Preview.java:448)
        at androidx.camera.core.Preview.invalidateMetadata(Preview.java:389)
        at androidx.camera.core.Preview.setTargetRotation(Preview.java:307)
        at com.android.example.cameraxbasic.fragments.CameraFragment$displayListener$1.onDisplayChanged(CameraFragment.kt:124)
        at android.hardware.display.DisplayManagerGlobal$DisplayListenerDelegate.handleMessage(DisplayManagerGlobal.java:633)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:193)
        at android.app.ActivityThread.main(ActivityThread.java:6669)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)

owahltinez, 2019-09-04:
Can you reproduce this consistently? I am unable to reproduce the issue on my Pixel device. Are you using the latest version of the sample?

owahltinez, 2019-10-02:
Please reopen if you are still experiencing this issue with the latest version of the library.

closed 2019-10-02 by owahltinez