android/camera-samples #53: CameraX - Preview

Wonderpol, 2019-08-15:
I use AutoFitPreviewBuilder in the same way as in sample but there is a problem on some devices e.g. on my Samsung s9 everything works fine but on my older(SM s6) this line: 
`val newViewFinderDimens = Size(right - left, bottom - top)` generate wrong values and there are ugly stripes(left and right side), the height is good. On my friend's phone it works well but there is another issue `updateTransform(viewFinder, rotation, it.textureSize, viewFinderDimens)` value `it.textureSize` is wrong. What can cose this problem if on some devices is everything all right. Could it be problem with CamerX API or the problem is on my side?

closed 2019-08-15 by Wonderpol

claireliu14, 2019-08-20:
Did you solve the problem ?

Wonderpol, 2019-08-20:
Hey, not at all. The problem disappears when app is made to full screen. As
I said not on every device there is a problem without full screen. But what
if I want to have status bar and good preview. I have already tried a lot
of things but without success. I think it can be connected with android
version because on phones e.g S9 S10+ works well but on e.g s6 not.

claireliu14, 2019-08-20:
Thanks for your reply. I met the same problem on s8, too. But after changed CameraX API version to 1.0.0-alpha03, it works well for preview.

Wonderpol, 2019-08-20:
Is your app full screen or not ?

claireliu14, 2019-08-21:
No, I have navigation bar and transparent status bar.

Wonderpol, 2019-08-21:
Ok  I'll try

Wonderpol, 2019-08-21:
Hello, could you help me to do it without transparent statu bar?
I have no idea why it works like that.

śr., 21 sie 2019 o 09:09 Jasiek Piaskowy <piaskowyjasiek@gmail.com>
napisał(a):

claireliu14, 2019-08-27:
@Wonderpol 
Maybe you can share your code and screenshot here.

VuManhHung97, 2019-10-07:
I had the same problem. Have you edited it yet?

VuManhHung97, 2019-10-08:
> No, I have navigation bar and transparent status bar.

How to make a preview not full screen?Can you share the code for me?