android/camera-samples #55: How to disable auto white balance or brightness in CameraX?

maxsdev, 2019-08-20:
When you change lighting of capturing subject, white balance on picture in preview changes. How to avoid this effect?

amartinz, 2019-08-29:
You can use the `Camera2Config.Extender` for now to set `CaptureRequest.CONTROL_AWB_MODE`  to `CaptureRequest.CONTROL_AWB_MODE_OFF` and other values to match your needs.

An example how to use Camera2Config.Extender:

```
// Set up the view finder use case to display camera preview
val viewFinderConfigBuilder = PreviewConfig.Builder().apply {
    setLensFacing(lensFacing)
    // We request aspect ratio but no resolution to let CameraX optimize our use cases
    setTargetAspectRatio(screenAspectRatio)
    // Set initial target rotation, we will have to call this again if rotation changes
    // during the lifecycle of this use case
    setTargetRotation(viewFinder.display.rotation)
}

// Create the extender and pass in the config builder we want to extend
val previewExtender = Camera2Config.Extender(viewFinderConfigBuilder)
// Turn AWB off
previewExtender.setCaptureRequestOption(CaptureRequest.CONTROL_AWB_MODE,
        CaptureRequest.CONTROL_AWB_MODE_OFF)

// Build your config as usual and create your wanted UseCase with it
val viewFinderConfig = viewFinderConfigBuilder.build()

// Use the auto-fit preview builder to automatically handle size and orientation changes
preview = AutoFitPreviewBuilder.build(viewFinderConfig, viewFinder)
```

maxsdev, 2019-08-29:
@amartinz Now I get an error

> IllegalArgumentException: Unsupported session configuration combination

owahltinez, 2019-09-04:
> @amartinz Now I get an error
> 
> > IllegalArgumentException: Unsupported session configuration combination

Keep in mind that once you use Camera2 extensions, you are on your own as far as checking if the device supports the requested frame configuration -- you might be getting errors like the one reported here. This might change in the future, but using the Camera2 extensions is currently the only way to change this setting.

closed 2019-09-04 by owahltinez

ratnesh77, 2019-10-07:
Keep in mind that once you use Camera2 extensions, you are on your own as far as checking if the device supports the requested frame configuration -- you might be getting errors like the one reported here. This might change in the future, but using the Camera2 extensions is currently the only way to change this setting.

Getting Same Issue::
IllegalArgumentException: Unsupported session configuration combination

Hey how to check device support or not, can you provide me any references

ratnesh77, 2019-10-07:
> Keep in mind that once you use Camera2 extensions, you are on your own as far as checking if the device supports the requested frame configuration -- you might be getting errors like the one [...]

Getting Same issue::
 IllegalArgumentException: Unsupported session configuration combination

How to check hardware support or not, can you give me any references

Coolnesss, 2019-10-18:
Any setting of `captureRequest.CONTROL_AWB_MODE` crashes to ` IllegalArgumentException: Unsupported session configuration combination` here as well. Surely atleast the basic settings, e.g `DAYLIGHT` are supported on all phones? Is this a bug?

IAn2018cs, 2019-12-14:
> Any setting of `captureRequest.CONTROL_AWB_MODE` crashes to ` IllegalArgumentException: Unsupported session configuration combination` here as well. Surely atleast the basic settings, e.g `DAYLIGHT` are supported on all phones? Is this a bug?

I found that the extension configuration was checked in androidx.camera.core.SessionConfig, and if the new configuration conflicts with the old configuration, it will throw Unsupported session configuration combination

lunayo, 2020-02-28:
I have the same problem of trying disabling the AWB using the method above, but it does seem that CameraX has validation and does not allow overriding of config that is controlled by CameraX. Instead of trying to disable the AWB using `captureRequest.CONTROL_AWB_MODE`, I just lock the AWB using `captureRequest.CONTROL_AWB_LOCK` to get the images with consistent white balance.

x64Dev, 2020-04-18:
Hi,
I'm also looking for a similar solution, in my case I want to change the brightness and contrast of the camera but unable to find a solution. Can any one please help me with this?

lunayo, 2020-04-18:
@x64Dev If you need to change the brightness and contrast and setting that is specifically related to `CaptureRequest` it would be better to use directly camera2 api instead of CameraX. Apparently CameraX has limitation on doing request customisation and it would need you to rebind the use case, causing camera to restart again. I have switched CameraX to camera2 since then. The CameraBasic is very good example of that.

x64Dev, 2020-04-18:
> @x64Dev If you need to change the brightness and contrast and setting that is specifically related to `CaptureRequest` it would be better to use directly camera2 api instead of CameraX. Apparently [...]

Thanks for the info. Why are they advertising using CameraX instead of previous Api if it's ain't ready/capable? Any way thanks for the suggestion. Let me check Camera2 then.

huntc, 2020-04-22:
> Keep in mind that once you use Camera2 extensions, you are on your own as far as checking if the device supports the requested frame configuration -- you might be getting errors like the one [...]

@owahltinez @amartinz  - I notice that the `PreviewConfig.Builder` no longer exists for CameraX beta3. Any update in terms of what must now be done to set the underlying Camera2 configuration?

lunayo, 2020-04-22:
> @owahltinez @amartinz - I notice that the `PreviewConfig.Builder` no longer exists for CameraX beta3. Any update in terms of what must now be done to set the underlying Camera2 configuration?

You could try this
`val analysisExtend = Camera2Interop.Extender(builder)
 analysisExtend.setCaptureRequestOption(CaptureRequest.CONTROL_AWB_LOCK,
 false)`