android/camera-samples #57: CameraX - Select cameraId

urunimi, 2019-09-01:
I saw that it provides very easy and convenient settings using [CameraX.LensFacing](https://developer.android.com/reference/kotlin/androidx/camera/core/CameraX.LensFacing.html)
However, it looks like developers can only select the direction the camera but can't select the index of them. Modern Android devices usually have more than one camera module on the back or front side. 
For example, camera2 supports this by providing this. [cameraId](https://developer.android.com/reference/android/hardware/camera2/CameraManager.html#openCamera(java.lang.String,%20android.hardware.camera2.CameraDevice.StateCallback,%20android.os.Handler))
Is or will be there a way to select one of them?

owahltinez, 2019-09-04:
Hi @urunimi. CameraX works at a different abstraction layer, above selecting a specific camera identifier. Instead, you request which lens orientation you want and which use cases you are trying to execute, and CameraX will pick the best camera that matches your desired configuration.

closed 2019-09-04 by owahltinez

yxchng, 2020-05-14:
@owahltinez When will we be able to select camera by ID? Especially when I have more than one cameras on the front or on the back? Is it going to be implemented in the future?

RainliFu, 2021-08-03:
```

   val cameraManager = getSystemService(Context.CAMERA_SERVICE) as CameraManager
   val  cameraIdList = cameraManager.cameraIdList
   val cameraSelector = getCameraSelector(cameraIdList[0])

@SuppressLint("UnsafeExperimentalUsageError", "UnsafeOptInUsageError", "RestrictedApi")
    private fun getCameraSelector(cameraId: String): CameraSelector {
        Timber.d("getCameraSelector $cameraId")
        return CameraSelector.Builder()
            .addCameraFilter { cameras ->
                Timber.d("addCameraFilter is ${cameras.size}")
                val result = cameras.filter {
                    cameraId == Camera2CameraInfo.from(it).cameraId
                }
                result
            }
            .build()
    }
```

I found this function to get CameraSelector by camera id .

zhaoguanao, 2022-01-15:
> 

Sry for disturbe, i was wondering what does "it" in "cameraId == Camera2CameraInfo.from(it)" mean? Is it the CameraInfo related to the camera we wants to use?

zhaoguanao, 2022-01-15:
Hi, @RainliFu, sry for disturbe, i was wondering what is "cameras" and "it" in your code? I'm confused