android/camera-samples #59: Knowing actual selected resolution before analyze is called for the first time?

sonicdebris, 2019-09-15:
I using the `ImageAnalysis` use case, and I need to initialize my analysis code with the image resolution (width and height).

Is there any way to get the resolution picked during camera initialization any time before the first call to `ImageAnalysis.Analyzer.analyze(...)` ?

owahltinez, 2019-09-27:
Good question. We checked with the product team working on CameraX and this is not currently possible today, but we are looking into potentially new APIs to enable developers to do this.

owahltinez, 2019-10-02:
Closing for now. Feel free to reach out to the team at https://g.co/camerax/developers if you want to start a discussion about this over there!

closed 2019-10-02 by owahltinez