android/camera-samples #60: CameraX chooses low resolution for given aspect ratio

Dem0n13, 2019-09-16:
I want to get the best quality for square aspect ratio and setup the next preview and capture configs for CameraX.

```
val SQUARE_ASPECT_RATIO = Rational(1, 1)

val previewConfig = PreviewConfig.Builder().apply {
    setTargetAspectRatio(SQUARE_ASPECT_RATIO)
    setTargetRotation(viewFinder.display.rotation)
}.build()
preview = Preview(previewConfig)

val imageCaptureConfig = ImageCaptureConfig.Builder().apply {
    setCaptureMode(ImageCapture.CaptureMode.MAX_QUALITY)
    setTargetAspectRatio(SQUARE_ASPECT_RATIO)
    setTargetRotation(viewFinder.display.rotation)
}.build()
imageCapture = ImageCapture(imageCaptureConfig)
```

CameraX.bindToLifecycle(this, preview, imageCapture)
CameraX chooses 352x288px resolution, so result image resolution equals 288x288px. From documentation it should be the highest resolution available for current device, but it is not.

Tested on emulator and Google Pixel 3.

owahltinez, 2019-09-27:
What resolutions are available in the devices you are testing this in? Asking for a squared aspect ratio will severely limit the resolutions that CameraX will attempt to choose from.

Giammaofwar, 2019-10-01:
Got the same problem, if I specify the aspect ratio CameraX will choose an incredible low resolution. Tried with more ratio on really good camera (Galaxy s10+)

owahltinez, 2019-10-02:
Please try again with the updated version of the sample since we just upgraded to `alpha05`. This problem should have been resolved now, feel free to reopen if that's not the case.

closed 2019-10-02 by owahltinez