android/camera-samples #61: java.lang.IllegalStateException: buffer must not be null

barryzzz, 2019-09-17:
I switched the phone camera on the samsung c5000(android 8.0) and it crashed

2019-09-17 11:35:40.534 32667-32690/com.android.example.cameraxbasic D/Camera: Opening camera: 0
2019-09-17 11:35:40.535 32667-32667/com.android.example.cameraxbasic E/BufferItemConsumer: [ImageReader-640x480f23m4-32667-11] Failed to release buffer: Unknown error -1 (1)
2019-09-17 11:35:40.536 32667-32690/com.android.example.cameraxbasic D/UseCaseAttachState: All use case: [androidx.camera.core.Preview-9fa4159c-cd82-4074-b27f-c75853c3fc55, androidx.camera.core.ImageCapture-23a0ebfe-621a-4e28-882c-f4b059c7c3f9, androidx.camera.core.ImageAnalysis-8162ef51-de08-4215-9036-284be2a7a5ec] for camera: 0
2019-09-17 11:35:40.536 32667-32690/com.android.example.cameraxbasic I/CameraManager: getCameraCharacteristics : cameraId = 0
2019-09-17 11:35:40.539 32667-32691/com.android.example.cameraxbasic E/AndroidRuntime: FATAL EXCEPTION: LuminosityAnalysis
    Process: com.android.example.cameraxbasic, PID: 32667
    java.lang.IllegalStateException: buffer must not be null
        at com.android.example.cameraxbasic.fragments.CameraFragment$LuminosityAnalyzer.analyze(CameraFragment.kt:459)
        at androidx.camera.core.ImageAnalysisNonBlockingCallback$1.run(ImageAnalysisNonBlockingCallback.java:150)
        at android.os.Handler.handleCallback(Handler.java:789)
        at android.os.Handler.dispatchMessage(Handler.java:98)
        at android.os.Looper.loop(Looper.java:164)
        at android.os.HandlerThread.run(HandlerThread.java:65)

owahltinez, 2019-09-27:
It looks like an issue with the library itself, not the sample. Can you please file a bug in the appropriate component of the Android Issue Tracker and reply here once the you have opened the issue there? It should be under https://issuetracker.google.com/components/618491

owahltinez, 2019-10-02:
We just upgraded the sample to the `alpha05` version of the CameraX library. If you can still reproduce this issue, please open a bug in the link I pasted above.

closed 2019-10-02 by owahltinez