android/camera-samples #62: Cannot swtch camera

jayeshbabuav, 2019-09-18:
When I switch the camera, app is getting stuck and this error is shown on the log:
`E/GLConsumer: [SurfaceTexture-0–6996–1] updateAndRelease: GLConsumer is not attached to an OpenGL ES context`

owahltinez, 2019-09-27:
What device are you seeing this in? Can you please file a bug in the appropriate component of the Android Issue Tracker and reply here once the you have opened the issue there? It should be under https://issuetracker.google.com/components/618491

owahltinez, 2019-10-02:
We just upgraded the sample to the `alpha05` version of the library. If you are still hitting this issue with the latest version of the sample, please open a bug using the link I pasted above.

closed 2019-10-02 by owahltinez