android/camera-samples #63: set fps for CameraX java code

elafefy11, 2019-09-19:
hello , 
i need to set camerax fps with java code  , 
i use CameraX in my app but the preview is very slow and blurring 
A lot of distortion in the picture
how i can set fps but use java code ?

owahltinez, 2019-09-27:
For the blurry preview, I think that you might be hitting this issue: https://issuetracker.google.com/issues/140194349

FPS should not be slow. What device are you seeing this in?

elafefy11, 2019-09-27:
thanks for reply ,
i use Samsung Galaxy S4 android lollipop 5 ,
and The current "camera-samples" works well on my device

owahltinez, 2019-10-02:
It's possible that you are hitting device limitations for concurrent streams. I would suggest disabling the `ImageAnalysis` use case to see if FPS improves.

owahltinez, 2019-11-29:
Closing due to inactivity

closed 2019-11-29 by owahltinez