android/camera-samples #67: How can I draw canvas on cameraTextureView

hajsf, 2019-09-25:
I tried below but nothing happened:
```kotlin
val paint = Paint().apply {
             isAntiAlias = true
             style = Paint.Style.STROKE
             color = Color.RED
            strokeWidth = 10f
}

var canvas = Canvas(bitmap)
canvas.drawRect(
           rect.x.toFloat(),
           rect.y.toFloat(), 
           rect.x.toFloat() + rect.width,
          rect.y.toFloat() + rect.height,
)

cameraTextureView.unlockCanvasAndPost(canvas)
```

owahltinez, 2019-09-27:
What sample are you trying to do this with? Camera2 or CameraX? My first guess would be that you are drawing on the TextureView but the camera keeps outputting new frames and drawing on top of your code, which is why you are not seeing any changes. You need to do the drawing in a different layer or implement some synchronization mechanism.

hajsf, 2019-09-27:
mm, could be your statement is correct @owahltinez 
> the camera keeps outputting new frames and drawing on top of your code
> You need to do the drawing in a different layer

Below what I did, appreciate your comments for optimization

I created an image view, with the same dimension, and location (constrains) of the cameeraX texture view, to act as overlay, and do all the drawings on it,  just need to be sure that this overlay is above not under the texture view, i.e. following it in the layout xml file.

```xml
<?xml version="1.0" encoding="utf-8"?>
<androidx.constraintlayout.widget.ConstraintLayout xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:app="http://schemas.android.com/apk/res-auto"
    xmlns:tools="http://schemas.android.com/tools"
    android:layout_width="match_parent"
    android:layout_height="match_parent"
    tools:context=".MainActivity">

    <TextureView
        android:id="@+id/view_finder"
        android:layout_width="wrap_content"
        android:layout_height="wrap_content"
        android:layout_marginTop="24dp"
        android:visibility="visible" />

    <ImageView
        android:id="@+id/imageView"
        android:layout_width="wrap_content"
        android:layout_height="wrap_content"
        android:contentDescription="@string/pic_desc"
        android:visibility="visible" />

</androidx.constraintlayout.widget.ConstraintLayout>
```
And the drawing code is:
```
lateinit var overlay: Bitmap
private fun startCamera() {
        val analyzerConfig = ImageAnalysisConfig.Builder().apply {
            setImageReaderMode(
                ImageAnalysis.ImageReaderMode.ACQUIRE_LATEST_IMAGE)
        }.build()

        val analyzerUseCase = ImageAnalysis(analyzerConfig).apply {
            analyzer = ImageAnalysis.Analyzer { image, rotationDegrees ->
                val bitmap = view_finder.bitmap ?: return@Analyzer
                overlay = Bitmap.createBitmap(bitmap.width, bitmap.height, Bitmap.Config.ARGB_8888)

                scope.launch(Dispatchers.Unconfined) {
                    val mat = Mat()
                    Utils.bitmapToMat(bitmap!!, mat)

                    val detectedFaces = FaceDetection.detectFaces(bitmap!!)
 
                   if (detectedFaces.toArray().isNotEmpty()) {
                       val paint = Paint().apply {
                           isAntiAlias = true
                           style = Paint.Style.STROKE
                           color = Color.RED
                           strokeWidth = 10f
                       }

                       for (rect in detectedFaces.toArray()) {
                           var canvas = Canvas(overlay)
                           canvas.drawRect(
                               rect.x.toFloat(), 
                               rect.y.toFloat(), 
                               rect.x.toFloat() + rect.width,
                               rect.y.toFloat() + rect.height,
                               paint
                           )
                           overlay?.let { Canvas(it) }?.apply {
                               canvas
                           }
                       }
                   }
                }

                runOnUiThread {
                    imageView.setImageBitmap(overlay)
                }
            }
        }
   CameraX.bindToLifecycle(this, preview, imageCapture, analyzerUseCase)
}
```

Below is an output of my app as of now showing this drawing:

[![enter image description here][1]][1]

  [1]: https://i.stack.imgur.com/Lp8QT.jpg

closed 2019-09-30 by hajsf