android/camera-samples #69: Error : No supported surface combination is found for camera device - Id : 0.  May be attempting to bind too many use cases.

vishalsgithub, 2019-09-27:
i just trying to record video with cameraX  usign below code 

~~~
mVideoFileSaver = new VideoFileSaver();
            VideoCaptureConfig config = new VideoCaptureConfig.Builder()
                    .setLensFacing(mCurrentCameraLensFacing)
                    .setTargetResolution(new Size(1280, 720))
                    //.setTargetName("VideoCapture")
                    .build();
            mVideoCapture = new VideoCapture(config);
            CameraX.bindToLifecycle(this,  mVideoCapture);
            //CameraX.bindToLifecycle(this, mPreview, mVideoCapture);
            mVideoCapture.startRecording(mVideoFileSaver.getNewVideoFile(),
                    mVideoFileSaver, new VideoCapture.Metadata());
~~~

and it's return error 

~~~
Error: No supported surface combination is found for the camera device - Id : 0.  May be attempting to bind too many use cases.
~~~

owahltinez, 2019-09-29:
Hi Vish, where are you encountering this issue? None of the samples of this repo use VideoCapture, since that is not even supported by the library.

vishalsgithub, 2019-09-30:
I'm trying to record video with cameraX, I found this code from StackOverflow and the answer was green marked I mean it's working code.but when I run this code it returns an error.

I have added below library in my project

~~~
implementation "androidx.camera:camera-core:1.0.0-alpha05"
implementation "androidx.camera:camera-camera2:1.0.0-alpha05"
~~~

I don't know VideoCapture is supported or not by this library

owahltinez, 2019-10-08:
VideoCapture is not yet supported by the library. Issues in this Github project are for problems related to the sample only. For issues with the library, there is the AOSP issue tracker: https://issuetracker.google.com/components/618491

closed 2019-10-08 by owahltinez