android/camera-samples #70: Audio time lag on video file

GRosay, 2019-09-27:
Hey there !
I've got an issue using the Java sample to record video/audio.

In the saved video file, there's a small gap between video and audio, something like half a second to a complete second (depending on the video duration).

How can I fix this bug ?

owahltinez, 2019-10-02:
Thanks for opening this issue. A couple of questions:

- Does this happen in the Kotlin version of the sample as well?
- What OS version is your Galaxy Note 8 running? I don't have one of those available, but I'll try to get my hands on one to see if I can reproduce this issue.

Device-specific problems are pretty hard to resolve, since it could be triggered by a number of things including device-specific bugs in their stack.

GRosay, 2019-10-02:
Thanks for coming back to me, I’ll try to give you as much information as you may need.

- That’s a good question, I’ll try with Kotlin ASAP and get back to you.
- The Galaxy Note 8 is running under Android 9.

owahltinez, 2020-01-28:
Closing due to inactivity, please reopen if it's still an issue and can provide further information

closed 2020-01-28 by owahltinez

DroidMagician, 2022-06-06:
Same issue I am getting 
After recording the video there is a half / 1-second delay in audio between the actually captured audio time and the preview video's audio time 
Using Samsung M30 Device and Android version is 10 
Please do let me know if more information required 
@owahltinez