android/camera-samples #73: Have problems with memory usage in Camera2BasicKotlin.

AntipovAlexander, 2019-09-30:
Have problems with memory usage in Camera2BasicKotlin.

When the app from Camera2BasicKotlin repository is running, I am facing with continually increasing memory usage.

GIF describing the issue is attached.

[image]

Is it a problem? What should I do with it?

Steps to reproduce:
1. Clone and build Camera2BasicKotlin
2. Deploy it to device
3. Run and give permissions for Camera
4. Just leave the app opened and nothing with it
5. Open profiler
6. Observe memory graph

Actual result: memory usage is continually increasing.

owahltinez, 2019-10-02:
Thank you for opening this issue!

> Is it a problem?

I will have to take a deeper look, but it definitely looks like a problem to me...

> What should I do with it?

Does it happen in the Java version of this sample as well? Another thing you can do is try to help us root cause the memory leak :)

owahltinez, 2019-10-02:
Also, thank you for attaching the GIF. It makes it **so much easier** to understand bug reports like this one!

AntipovAlexander, 2019-10-03:
Thank you for your reply!

> Does it happen in the Java version of this sample as well?

Yes, the Java version has the same problem.

Also, I have checked other camera samples: CameraX and NDK sample, and got interesting results.

It seems like CameraX forces invoke of garbage collector under the hood, pay attention to the trash box icon:
[image]

And NDK sample, perfect straight line :)
[image]

owahltinez, 2019-10-18:
Hey @AntipovAlexander sorry for the late reply. What specific device are you seeing this in? I spent some time investigating this issue, here is what I think is happening:

Memory usage growth is only happening from the **native** allocations. I'm 99% sure this means that the issue comes from the HAL (i.e. the device drivers) since we are not calling any native code and we are not creating objects in these samples while the preview is just running.

Further, on my Pixel 3 XL device, I see the garbage collector kicking in and reducing that memory usage. So it's likely that, if memory in your device runs low, the garbage collector will kick in as well. You can test this out by forcing GC in the profiler (right click > Force garbage collection) or potentially even by calling `System.gc()` in your code -- although I do not recommend that approach in a production app.

I have also tried one of our new samples, [Camera2SlowMotion](https://github.com/android/camera-samples/tree/master/Camera2SlowMotion) and the GC kicks in at more regular intervals, with native memory usage ranging pretty stable between 10MB and 20MB:

[image]

hikkidev, 2019-10-31:
[image]
[image]

SumeetGohil, 2019-11-04:
Samsung Galaxy J2 Prime

[image]

AntipovAlexander, 2019-11-04:
Hi @owahltinez 

I have tested this issue on two physical devices: Huawei P30 Pro (8gb ram), OnePlus 5 (6gb ram) and one emulator (built-in AS, API 28). I was always running into this issue on all my devices. 

Unfortunately, I don't have any Pixel devices, so I can't test it like you.

I have already tried forcing GC - it works, but it's not a solution for us :) And as I said, this issue happens only Camera2Basic(Kotlin/Java) samples, but on CameraX sample GC works the same way, as you described.

owahltinez, 2019-11-06:
I think forcing GC is not a solution to the problem, it simply indicates that, if there's memory pressure, the system will be able to free up that memory. The way CameraX operates might be triggering the GC because of some other condition.

What is the problem that you are having, exactly? High memory usage shouldn't affect your app users, since memory will be freed from the native camera stack whenever other processes need that memory.

owahltinez, 2019-12-23:
Closing due to inactivity. Please reopen if you are still experiencing issues with this. You can also start a thread on the [developer forums](g.co/camerax/developers) if you wish!

closed 2019-12-23 by owahltinez

liungkejin, 2020-04-12:
> Hi @owahltinez
>
> I have tested this issue on two physical devices: Huawei P30 Pro (8gb ram), OnePlus 5 (6gb ram) and one emulator (built-in AS, API 28). I was always running into this issue on all my [...]

Hi @AntipovAlexander 
You're right! I tried many solutions, the only solution is forcing GC, but not call `System.gc()`
```java
    private byte [] fuckingData = null;
    private void fixFuckingCam2OOM() {
        if (fuckingData == null) {
            fuckingData = new byte[1024 * 1024];
        } else {
            fuckingData = null;
        }
    }
```
I call this method every camera frame! this will cause system gc per second!
My devices is `Nokia x7 / 9.0`

yanglichuan, 2020-11-27:
i have the same question