android/camera-samples #74: CameraX wrong image orientation

Holoceo, 2019-10-01:
When I take a picture in CameraX sample with LensFacing.FRONT it is displayed correctly in photo screen, but other apps shows it rotated by 90 degrees clockwise.

One Plus 5, Android 9.0 (OxygenOS 9.0.8)

owahltinez, 2019-10-02:
It sounds like the other apps are not reading the EXIF metadata properly. Do you know what is the package name of those other apps? Is there any way for you to submit a bug report to them?

Holoceo, 2019-10-02:
Google Drive displays this image even upside down. However preview is good.

owahltinez, 2019-10-02:
So the preview in the Google Drive app appears correct, but after you open the file it shows upside down? That's a good one. What version of the app is this happening on?

Holoceo, 2019-10-02:
Yes. The version is 2.19.352.03.40

owahltinez, 2019-10-02:
You should be able to file a bug for Google Drive by going to the hamburger menu > Help & feedback > Send feedback.

Developers can modify the images if they choose to in their apps, CameraX only adds the EXIF metadata and tries to stay away from modifying the pixels of the camera frames.

Holoceo, 2019-10-02:
Ok, but looks like there are no apps that honor EXIF. Even instagram, whatsapp, telegram, file chooser, gmail attachment preview... I haven't found an app which displays this picture correctly.

owahltinez, 2019-10-02:
This is good feedback, and maybe at some point we should consider that maybe it's us who are doing it wrong and not *everyone else*. I encourage you to start a conversation with the team directly at https://g.co/camerax/developers.

It doesn't help that Android's Bitmap factory libraries all ignore EXIF metadata and developers have to load yet another library to do the right thing. Developers that use `Glide` should be getting the images with the correct orientation by default.

IhwanID, 2019-10-03:
I have some issue.

inludese, 2019-10-28:
I have the same issue.
Preview looks good for LensFacing.FRONT and LensFacing.Back, but all stored images have landscape orientation if I open it in Android Studio. 
After I stored images on Mac, every image have portrait orientation. ( portrait or landscape orientation of device did not affect output image at all)

Device: Pixel 
OS Version: Android 10

owahltinez, 2019-10-29:
Thanks @Holoceo for opening up the discussion at the forums. For everyone's reference, here's the thread where you can engage directly with the engineering team and ask for a fix: https://groups.google.com/a/android.com/forum/#!topic/camerax-developers/NqZG8Bt4iOg

I'll keep this issue open so people can find it and get some context, but currently it is not actionable since we don't want to rotate images in the sample -- at least for the time being.

owahltinez, 2019-11-21:
Actually, this issue will show up from Google Search even if it's closed. Closing!

closed 2019-11-21 by owahltinez