android/camera-samples #75: Lens Switch Issue

tehilaschechter, 2019-10-03:
When user taps the "switch lens" button, the preview freezes but does not switch. Then, when the app is closed and re-opened, the camera has now switched lenses (front to back or vice versa). What can I do so that the camera will switch lenses on demand (both in preview and in capture)?

package com.example.camerax

import android.Manifest
import android.annotation.SuppressLint
import android.content.Intent
import android.graphics.BitmapFactory
import android.graphics.Matrix
import android.net.Uri
import android.os.Build
import android.os.Bundle
import android.os.Environment
import android.util.DisplayMetrics
import android.util.Rational
import android.util.Size
import android.view.Surface
import android.widget.Button
import android.widget.Toast
import androidx.appcompat.app.AppCompatActivity
import androidx.camera.core.*
import kotlinx.android.synthetic.main.activity_main.*
import java.io.File
import java.net.URI

class MainActivity: AppCompatActivity() {

    private var lensFacing = CameraX.LensFacing.FRONT //default to front camera
    private val TAG = "MainActivity"

    override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)
        setContentView(R.layout.activity_main)

        //TODO: ask for permissions before starting the app, not after. Otherwise it crashes until given

        //check permissions
        if (Build.VERSION.SDK_INT >= 23) { // apps with higher sdk can revoke permissions at any time
            requestPermissions(
                arrayOf(
                    Manifest.permission.CAMERA,
                    Manifest.permission.WRITE_EXTERNAL_STORAGE
                ), 2
            ) //every time we want to use these permissions, we must check to ensure that we have them
        }

        texture.post { startCamera() }

        // Every time the provided texture view changes, recompute layout
        texture.addOnLayoutChangeListener { _, _, _, _, _, _, _, _, _ ->
            updateTransform()
        }
    }

    @SuppressLint("RestrictedApi")
    private fun startCamera() {
        val metrics = DisplayMetrics().also { texture.display.getRealMetrics(it) }
        val screenSize = Size(metrics.widthPixels, metrics.heightPixels)
        val screenAspectRatio = Rational(metrics.widthPixels, metrics.heightPixels)

        val previewConfig = PreviewConfig.Builder().apply {
            setLensFacing(lensFacing)
            setTargetResolution(screenSize)
            setTargetAspectRatio(screenAspectRatio)
            setTargetRotation(windowManager.defaultDisplay.rotation)
            setTargetRotation(texture.display.rotation)
        }.build()

        val preview = Preview(previewConfig)
        preview.setOnPreviewOutputUpdateListener {
            texture.surfaceTexture = it.surfaceTexture
            updateTransform()
        }

        // Create configuration object for the image capture use case
        val imageCaptureConfig = ImageCaptureConfig.Builder()
            .apply {
                setLensFacing(lensFacing)
                setTargetAspectRatio(screenAspectRatio)
                setTargetRotation(texture.display.rotation)
                setCaptureMode(ImageCapture.CaptureMode.MAX_QUALITY)
            }.build()

        // Build the image capture use case and attach button click listener
        val imageCapture = ImageCapture(imageCaptureConfig)
        btn_take_picture.setOnClickListener {

            val filename = "CameraX${System.currentTimeMillis()}.jpg"
            val sd = Environment.getExternalStorageDirectory()
            val file = File(sd, filename)

            imageCapture.takePicture(file,
                object : ImageCapture.OnImageSavedListener {
                    override fun onError(
                        imageCaptureError: ImageCapture.ImageCaptureError,
                        message: String,
                        cause: Throwable?
                    ) {
                        val msg = "Photo capture failed: $message"
                        Toast.makeText(baseContext, msg, Toast.LENGTH_LONG).show() //TODO change to LENGTH_SHORT

                    }

                    override fun onImageSaved(file: File) {
                        val photoUri = Uri.fromFile(File("\${file.absolutePath}"))
                        val evaluateImageIntent = Intent(this@MainActivity, EvaluateImage::class.java)

                        evaluateImageIntent.putExtra("imagePath", file.absolutePath)
                        startActivity(evaluateImageIntent)

                        CameraX.unbindAll()

                        val msg = "Photo capture successfully: ${file.absolutePath}" //TODO: DELETE
                        Toast.makeText(baseContext, msg, Toast.LENGTH_SHORT).show() // TODO: DELETE
                    }
                })

        }

        // Listener for button used to switch cameras
        btnSwitchLens.setOnClickListener {
            if (CameraX.LensFacing.FRONT == lensFacing) {
                val msg = "called switch lens" //TODO: DELETE
                Toast.makeText(baseContext, msg, Toast.LENGTH_SHORT).show() // TODO: DELETE
                lensFacing = CameraX.LensFacing.BACK

            } else {
                lensFacing = CameraX.LensFacing.FRONT
            }

            try {
                // Only bind use cases if we can query a camera with this orientation
                CameraX.getCameraWithLensFacing(lensFacing)

                // Unbind all use cases and bind them again with the new lens facing configuration
                CameraX.unbindAll()
                startCamera()

            } catch (exc: Exception) {

            }
        }

        CameraX.bindToLifecycle(this, preview, imageCapture)
    }

    private fun updateTransform() {
        val matrix = Matrix()
        val centerX = texture.width / 2f
        val centerY = texture.height / 2f

        val rotationDegrees = when (texture.display.rotation) {
            Surface.ROTATION_0 -> 0
            Surface.ROTATION_90 -> 90
            Surface.ROTATION_180 -> 180
            Surface.ROTATION_270 -> 270
            else -> return
        }
        matrix.postRotate(-rotationDegrees.toFloat(), centerX, centerY)
        texture.setTransform(matrix)
    }
}

owahltinez, 2019-10-03:
Hello, it looks like you are modifying the sample code. Does the issue occur when you use the sample as-is? What is the exact device model that you are using?

raranov, 2019-10-03:
I am using a Samsung S8

raranov, 2019-10-06:
What can I do to get rid of this bug?

owahltinez, 2019-10-08:
I am not able to reproduce this bug using the sample code as-is. If you are still experiencing this issue, it must be either with your code or with the library. Can you please file a bug in the appropriate component of the Android Issue Tracker? It should be under https://issuetracker.google.com/components/618491

closed 2019-10-08 by owahltinez